SpikeInterface/spikeinterface #466: TridesclousSortingExtractor requires write permission

JuliaSprenger, 2022-01-18:
When reading a tridesclous sorting of files with read-only permission the TridesclousSortingExtractor fails due to a PermissionError.
```
    sorting = TridesclousSortingExtractor(folder_path=output_folder)
../../extractors/tridesclousextractors.py:22: in __init__
    dataio = tdc.DataIO(str(tdc_folder))
../../../../../site-packages/tridesclous/dataio.py:127: in __init__
    self._open_processed_data()
../../../../../site-packages/tridesclous/dataio.py:495: in _open_processed_data
    self.arrays[chan_grp][i].load_if_exists(name)
../../../../../site-packages/tridesclous/iotools.py:294: in load_if_exists
    arr = np.memmap(self._fname(name), dtype=dtype, mode='r+')
PermissionError: [Errno 13] Permission denied: 'sorters/tests/sorting_tdc_docker/channel_group_0/segment_0/processed_signals.raw'
```

I don't think the writing permission should be required for extracting the sorting information. Maybe it would be good to introduce a reading mode flag here?

JuliaSprenger, 2022-01-18:
It is the output of a sorting run in a docker container. But I could also imagine sorting results being read-only for some users on purpose.

alejoe91, 2022-04-01:
Duplicate of https://github.com/SpikeInterface/spikeinterface/issues/474

closed 2022-04-01 by alejoe91