SpikeInterface/spikeinterface #469: Error message from spykingcircusextractors.py

MNS57, 2022-01-19:
Dear Buccino

From some point (initially it was fine), the spikeinterface started to say this error message.
Could you give any advice to solve it?

File "/home/nom/.pyenv/versions/anaconda3-2018.12/lib/python3.7/site-packages/spikeinterface/extractors/spykingcircusextractors.py", line 42, in __init__
    assert isinstance(parent_folder, Path) and isinstance(result_folder, Path), "Not a valid spyking circus folder"

AssertionError: Not a valid spyking circus folder

alejoe91, 2022-01-19:
Hi, or seems that spike sorting has failed. Can you attach the spikeinterface_log.json in the output folder?

MNS57, 2022-01-19:
Thanks! -- This is it.

{
    "sorter_name": "spykingcircus",
    "sorter_version": "1.1.0",
    "datetime": "2022-01-19T13:46:37.076656",
    "runtime_trace": [
        "",
        "##################################################################",
        "#####             Welcome to the SpyKING CIRCUS              #####",
        "#####                        (1.1.0)                         #####",
        "#####             Written by P.Yger and O.Marre              #####",
        "##################################################################",
        "",
        "",
        "File          : /home/MSNNlab/Applications/python_home/211125_spikeinterface/spykingcircus_output/recording.npy",
        "Steps         : filtering, whitening, clustering, fitting, merging",
        "Number of CPU : 12/24",
        "Parallel HDF5 : False",
        "Shared memory : True",
        "Hostfile      : /home/nom/spyking-circus/circus.hosts",
        "",
        "##################################################################",
        "",
        "",
        "-------------------------  Informations  -------------------------",
        "| Number of recorded channels : 64",
        "| Number of analyzed channels : 64",
        "| File format                 : NUMPY",
        "| Data type                   : float32",
        "| Sampling rate               : 20000 Hz",
        "| Duration of the recording   : 30 min 42 s 80 ms",
        "| Width of the templates      : 3 ms",
        "| Spatial radius considered   : 100 um",
        "| Threshold crossing          : negative",
        "------------------------------------------------------------------",
        "-------------------------  Informations  -------------------------",
        "| Using only 12 out of 24 local CPUs available (-c to change)",
        "------------------------------------------------------------------",
        "----------------------------  Error  -----------------------------",
        "| SpyKING CIRCUS does not work with OPENMPI >= 3.0",
        "| Consider downgrading or switching to MPICH",
        "------------------------------------------------------------------"
    ],
    "error": false,
    "run_time": 1.1663585589267313
}

MNS57, 2022-01-19:
Oh... The error is obvious... we check this point. 

"| SpyKING CIRCUS does not work with OPENMPI >= 3.0",
"| Consider downgrading or switching to MPICH",

alejoe91, 2022-01-19:
When something's off with spike sorting, always look at the log file! 😊

MNS57, 2022-01-19:
We appreciate the $100 million comment.

closed 2022-01-19 by alejoe91

phanhuynh, 2022-06-22:
Hi team, 
I got a similar error, but the log file was less helpful (attached)
[spikeinterface_log (2).txt](https://github.com/SpikeInterface/spikeinterface/files/8956445/spikeinterface_log.2.txt).
Should I remove the spyking circus version number (1.0.7)?  Here's the error message:
```
Docker: pulling image spikeinterface/spyking-circus-base:1.0.7
Starting container
Installing spikeinterface from sources in spikeinterface/spyking-circus-base:1.0.7
Installing dev spikeinterface from remote repository
Running spykingcircus sorter inside spikeinterface/spyking-circus-base:1.0.7
Stopping container
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Input In [2], in <cell line: 10>()
      1 ### Option 1: Run local sorter (not working) ##################
      2 # start_time = time.time()
      3 # sorting_SC = ss.run_sorter('spykingcircus', recording_saved, output_folder= '_output_SC', detect_threshold=threshold )
   (...)
      7 
      8 ### Option 2: run sorter directly from docker (preferred) ###
      9 start_time = time.time()
---> 10 sorting_SC = ss.run_sorter(
     11     'spykingcircus',
     12     recording_saved,
     13     output_folder= '_output_SC',
     14     docker_image="spikeinterface/spyking-circus-base:1.0.7",
     15     verbose=True,
     16 )
     17 print("--- _output_SC took %s seconds ---" % (time.time() - start_time))
     18 print(f'Spyking found {len(sorting_SC.get_unit_ids())} units')

File ~\Anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:124, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    122         else:
    123             container_image = singularity_image
--> 124     return run_sorter_container(
    125         container_image=container_image,
    126         mode=mode,
    127         **common_kwargs,
    128     )
    130 return run_sorter_local(**common_kwargs)

File ~\Anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:528, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    526 else:
    527     if with_output:
--> 528         sorting = SorterClass.get_result_from_folder(output_folder)
    530 if delete_output_folder:
    531     shutil.rmtree(output_folder)

File ~\Anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\basesorter.py:281, in BaseSorter.get_result_from_folder(cls, output_folder)
    277 if bool(log['error']):
    278     raise SpikeSortingError(
    279         "Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
--> 281 sorting = cls._get_result_from_folder(output_folder)
    283 recording = load_extractor(output_folder / 'spikeinterface_recording.json')
    284 if recording is not None:
    285     # can be None when not dumpable

File ~\Anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\spyking_circus\spyking_circus.py:169, in SpykingcircusSorter._get_result_from_folder(cls, output_folder)
    167 @classmethod
    168 def _get_result_from_folder(cls, output_folder):
--> 169     sorting = SpykingCircusSortingExtractor(folder_path=Path(output_folder) / 'recording')
    170     return sorting

File ~\Anaconda3\envs\si_env\lib\site-packages\spikeinterface\extractors\spykingcircusextractors.py:54, in SpykingCircusSortingExtractor.__init__(self, folder_path)
     52         break
     53 if results is None:
---> 54     raise Exception(spykingcircus_folder, " is not a spyking circus folder")
     56 # load params
     57 sample_rate = None

Exception: (WindowsPath('C:/Users/huynh/spike_epsztein/proj_louis/_output_SC/recording'), ' is not a spyking circus folder')

```

alejoe91, 2022-06-22:
Unfortunately this means that SC didn't finish correctly. Can you check the log file in the output folder?

phanhuynh, 2022-06-22:
Thanks for the speedy response :)
I attached the log file, but here's the key part:
        "===================================================================================",
        "=   BAD TERMINATION OF ONE OF YOUR APPLICATION PROCESSES",
        "=   PID 283 RUNNING AT ac12646857e6",
        "=   EXIT CODE: 7",
        "=   CLEANING UP REMAINING PROCESSES",
        "=   YOU CAN IGNORE THE BELOW CLEANUP MESSAGES",
        "===================================================================================",
        "YOUR APPLICATION TERMINATED WITH THE EXIT STRING: Bus error (signal 7)",
        "This typically refers to a problem with your application.",
        "Please see the FAQ page for debugging suggestions",
        "----------------------------  Error  -----------------------------",
        "| Step \"clustering\" failed for reason Command '['mpiexec', '-np', '18', 'spyking-circus-subtask', 'clustering', '/Users/huynh/spike_epsztein/proj_louis/_output_SC/recording.npy', '18', '0', 'False', 'False']' returned non-zero exit status 7.!",
        "------------------------------------------------------------------"
    ],
    "error": false,
    "run_time": 8872.105330819
}

alejoe91, 2022-06-23:
So, that's a version problem then! the newest tag is in fact 1.1,0 (see [here](https://hub.docker.com/repository/docker/spikeinterface/spyking-circus-base))