SpikeInterface/spikeinterface #473: Portability issue: hard-coded paths used for WaveformExtractor.load_from_folder

miketrumpis, 2022-01-21:
I believe this is a distinct issue, although it is at least a cousin to #422

The scenario is using spikeinterface 0.92.0 and mountainsort 1.0.0. Following a successful first-time run of these commands, the waveform extract directory is created.

```
sorting_MS4 = ss.run_mountainsort4(r, output_folder=output_dir, verbose=False, **params)
we = si.extract_waveforms(r, sorting_MS4, output_dir + '_waveforms', load_if_exists=True, ms_before=1, ms_after=2, 
                          max_spikes_per_unit=500, n_jobs=1, chunk_size=30000)
```

The issue here is that the `recording.json` and the `sorting.json` files in the extraction directory refer to full paths for the raw data and the sorting results, respectively. The result is that using the sorting/extraction results are not portable, and `extract_waveforms` needs to be run in the exact original environment.

I've seen this in two cases:

* moving sorting & extraction results to a different machine (without raw data)
* (example in traceback) trying to load a `WaveformExtractor` in a new python process or Docker container after sorting from data saved to a temporary file (e.g. after `NumpyRecording.save`)

Secondarily, does `WaveformExtractor` need a path on the raw data at all?

Right now I need to shim the json files before I can load the results. I was digging a little before I got busy, but maybe relative paths are supported?

traceback (note that the `None` argument for recording makes no difference)

```python
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
/tmp/ipykernel_116/923429382.py in <module>
----> 1 we = si.extract_waveforms(None, sorting_MS4, output_dir + '_waveforms', load_if_exists=True, ms_before=1, ms_after=2, 
      2                           max_spikes_per_unit=500, n_jobs=20, chunk_size=30000)

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py in extract_waveforms(recording, sorting, folder, load_if_exists, precompute_template, ms_before, ms_after, max_spikes_per_unit, overwrite, return_scaled, dtype, **job_kwargs)
    683         shutil.rmtree(folder)
    684     if load_if_exists and folder.is_dir():
--> 685         we = WaveformExtractor.load_from_folder(folder)
    686     else:
    687         we = WaveformExtractor.create(recording, sorting, folder)

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py in load_from_folder(cls, folder)
     90         folder = Path(folder)
     91         assert folder.is_dir(), f'This folder do not exists {folder}'
---> 92         recording = load_extractor(folder / 'recording.json')
     93         sorting = load_extractor(folder / 'sorting.json')
     94         we = cls(recording, sorting, folder)

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict, base_folder)
    755         return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    756     else:
--> 757         return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
    758 
    759 

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/base.py in load(file_path, base_folder)
    472             else:
    473                 raise ValueError(f'Impossible to load {file_path}')
--> 474             extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
    475             return extractor
    476 

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/base.py in from_dict(d, base_folder)
    301             assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    302             d = _make_paths_absolute(d, base_folder)
--> 303         extractor = _load_extractor_from_dict(d)
    304         folder_metadata = d.get('folder_metadata', None)
    305         if folder_metadata is not None:

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    684 
    685         if isinstance(v, dict) and is_dict_extractor(v):
--> 686             kwargs[k] = _load_extractor_from_dict(v)
    687 
    688             # handle list of extractors list

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    684 
    685         if isinstance(v, dict) and is_dict_extractor(v):
--> 686             kwargs[k] = _load_extractor_from_dict(v)
    687 
    688             # handle list of extractors list

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/base.py in _load_extractor_from_dict(dic)
    701 
    702     # instantiate extrator object
--> 703     extractor = cls(**kwargs)
    704 
    705     extractor._annotations.update(dic['annotations'])

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/binaryrecordingextractor.py in __init__(self, file_paths, sampling_frequency, num_chan, dtype, t_starts, channel_ids, time_axis, file_offset, gain_to_uV, offset_to_uV, is_filtered)
     81             else:
     82                 t_start = t_starts[i]
---> 83             rec_segment = BinaryRecordingSegment(datfile, sampling_frequency, t_start, num_chan, dtype, time_axis, file_offset)
     84             self.add_recording_segment(rec_segment)
     85 

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/binaryrecordingextractor.py in __init__(self, datfile, sampling_frequency, t_start, num_chan, dtype, time_axis, file_offset)
    127     def __init__(self, datfile, sampling_frequency, t_start, num_chan, dtype, time_axis, file_offset):
    128         BaseRecordingSegment.__init__(self, sampling_frequency=sampling_frequency, t_start=t_start)
--> 129         self._timeseries = read_binary_recording(datfile, num_chan, dtype, time_axis, file_offset)
    130 
    131     def get_num_samples(self) -> int:

/opt/conda/envs/msort/lib/python3.8/site-packages/spikeinterface/core/core_tools.py in read_binary_recording(file, num_chan, dtype, time_axis, offset)
    138     '''
    139     num_chan = int(num_chan)
--> 140     with Path(file).open() as f:
    141         nsamples = (os.fstat(f.fileno()).st_size - offset) // (num_chan * np.dtype(dtype).itemsize)
    142     if time_axis == 0:

/opt/conda/envs/msort/lib/python3.8/pathlib.py in open(self, mode, buffering, encoding, errors, newline)
   1220         if self._closed:
   1221             self._raise_closed()
-> 1222         return io.open(self, mode, buffering, encoding, errors, newline,
   1223                        opener=self._opener)
   1224 

/opt/conda/envs/msort/lib/python3.8/pathlib.py in _opener(self, name, flags, mode)
   1076     def _opener(self, name, flags, mode=0o666):
   1077         # A stub for the opener argument to built-in open()
-> 1078         return self._accessor.open(self, flags, mode)
   1079 
   1080     def _raw_open(self, flags, mode=0o777):

FileNotFoundError: [Errno 2] No such file or directory: '/tmp/spikeinterface_cache/tmpviuk8wbl/OCJYFCHA/traces_cached_seg0.raw'
```

samuelgarcia, 2022-01-21:
We norammly have machanism for relative path when do `.save()` for rec or sorting.

Maybe we haven't done this for waveform extarctor. We should do it.

@alejoe91 : could have a look ? and add the `relative_to=` option in waveform extractor dump ?

alejoe91, 2022-02-02:
@miketrumpis can you test this PR https://github.com/SpikeInterface/spikeinterface/pull/485?

You can now extract waveforms with the `use_relative_path=True` argument so that recording and sorting paths are relative to the waveform folder paths (provided they are on the same drive)

closed 2022-02-02 by samuelgarcia

miketrumpis, 2022-02-10:
I've since realized that I can recreate the waveform extractor with the basic constructor, so this feedback is only relevant in terms of approaching through the convenient `si.extract_waveforms(..., load_if_exists=True)`...

This part is great -- the sorter & extractor outputs can be moved in tandem.

[image]

And this part is a bit of a problem, I think. If I call that method later or on a different system, then the recording in the temp file will never be found. 

[image]

I would advocate that 

i. perhaps the load_from_folder class method should also accept a recording, if the recording has been re-created
ii. if the recording is not given, and the recording path is invalid, then "you get what you get" in terms of the saved waveforms.

https://github.com/SpikeInterface/spikeinterface/blob/202741a51110b64f64ef55f0bc7a26bd248aef0d/spikeinterface/core/waveform_extractor.py#L873-L874

miketrumpis, 2022-02-10:
Relatedly, it looks like in that commit and in 0.93.0 `BaseSorter.get_result_from_folder` is also now trying to reload the recording. Of course, the same temp file portability problem applies.