SpikeInterface/spikeinterface #474: Failed to run singularity and docker image without installing respective spikesorting package

caniko, 2022-01-24:
I am on Linux, Python 3.9, nvidia gtx 1650.

I couldn't run `tridesclous` without installing the package even though I was using docker/singularity image (tried with both). I want to avoid installing the package, and run on containers.

The tutorial ran fine till this point. Including `herdingspikes`, which had a succesful run (without docker/singularity).

Snippet that raises error:
``` python
sorting_TDC = ss.run_tridesclous(
    recording=recording_preprocessed,
    output_folder="tridesclous_output",
    singularity_image="spikeinterface/tridesclous-base:1.6.4"
)
```
Output:
``` shell
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Input In [9], in <module>
----> 1 sorting_TDC = ss.run_tridesclous(recording=recording_preprocessed, output_folder="tridesclous_output", singularity_image="spikeinterface/tridesclous-base:1.6.4")

File ~/.pyenv/versions/3.9.9/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:328, in run_tridesclous(*args, **kwargs)
    327 def run_tridesclous(*args, **kwargs):
--> 328     return run_sorter('tridesclous', *args, **kwargs)

File ~/.pyenv/versions/3.9.9/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:57, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, with_output, **sorter_params)
     50 """
     51 Generic function to run a sorter via function approach.
     52 
     53 >>> sorting = run_sorter('tridesclous', recording)
     54 """ + _common_param_doc
     56 if docker_image is None:
---> 57     sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     58                                remove_existing_folder=remove_existing_folder,
     59                                delete_output_folder=delete_output_folder,
     60                                verbose=verbose, raise_error=raise_error, with_output=with_output,
     61                                **sorter_params)
     62 else:
     63     sorting = run_sorter_docker(sorter_name, recording, docker_image, output_folder=output_folder,
     64                                 remove_existing_folder=remove_existing_folder,
     65                                 delete_output_folder=delete_output_folder,
     66                                 verbose=verbose, raise_error=raise_error,
     67                                 with_output=with_output, **sorter_params)

File ~/.pyenv/versions/3.9.9/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:81, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     78 SorterClass = sorter_dict[sorter_name]
     80 # only classmethod call not instance (stateless at instance level but state is in folder)
---> 81 output_folder = SorterClass.initialize_folder(
     82     recording, output_folder, verbose, remove_existing_folder)
     83 SorterClass.set_params_to_folder(
     84     recording, output_folder, sorter_params, verbose)
     85 SorterClass.setup_recording(recording, output_folder, verbose=verbose)

File ~/.pyenv/versions/3.9.9/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py:81, in BaseSorter.initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
     76 @classmethod
     77 def initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder):
     78 
     79     # installed ?
     80     if not cls.is_installed():
---> 81         raise Exception(f"The sorter {cls.sorter_name} is not installed."
     82                         f"Please install it with:  \n{cls.installation_mesg} ")
     84     if not isinstance(recording, BaseRecording):
     85         raise ValueError('recording must be a Recording!!')

Exception: The sorter tridesclous is not installed.Please install it with:  

To use Tridesclous run:

       >>> pip install tridesclous

    More information on tridesclous at:
      * https://github.com/tridesclous/tridesclous
      * https://tridesclous.readthedocs.io
     
```

alejoe91, 2022-01-24:
Hi @caniko!

Nice to hear from you! This happens only with TDC, because the extractor relies on the TDC package itself to load the spike sorting output. See these two issues (https://github.com/SpikeInterface/spikeinterface/issues/455 and https://github.com/SpikeInterface/spikeinterface/issues/466) for a discussiona about this. 

For other sorters, everything should work fine ;)

caniko, 2022-01-24:
Will keep this in mind, thank you!

alejoe91, 2022-08-01:
implemented

closed 2022-08-01 by alejoe91