SpikeInterface/spikeinterface #482: General issues with the easy install in Ubuntu 20.04

matiasandina, 2022-01-27:
Hello,
I am sorry if you prefer to chunk this in multiple issues, I thought it would be convenient to go at it all together, since I have solved some of them. I am happy to chunk it and add some suggestions/PR if you want me to.

### First check

I am trying to follow the install in `Ubuntu 20.04`. I created an environment using `conda` and the windows `.yml` file provided for the easy installations. I did this because I ran into some of this issues while installing in Windows (currently the only machine where we have a GPU with Kilosort) and I wanted to see if I could go all the way through in my linux machine. Running the python script to check the install I got:

```
write_binary_recording with n_jobs 1  chunk_size None
Import spikeinterface ...OK
Import spikeinterface.full ...OK
Run tridesclous ...OK
RUNNING SHELL SCRIPT: spykingcircus_output/run_spykingcircus.sh
Run spykingcircus ...OK
Run herdingspikes ...Fail
Open spikeinterface-gui ...Fail
Setting 'return_scaled' to False
Export to phy ...OK
Traceback (most recent call last):
  File "/home/matias/miniconda3/envs/si_env/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/home/matias/miniconda3/envs/si_env/lib/python3.8/site-packages/phy/apps/__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "/home/matias/miniconda3/envs/si_env/lib/python3.8/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import (
  File "/home/matias/miniconda3/envs/si_env/lib/python3.8/site-packages/phy/gui/qt.py", line 41, in <module>
    from PyQt5.QtWebEngineWidgets import (QWebEngineView,  # noqa
ImportError: /home/matias/miniconda3/envs/si_env/lib/python3.8/site-packages/PyQt5/../../../libQt5Network.so.5: undefined symbol: _ZN15QIPAddressUtils8toStringER7QStringPh, version Qt_5
Open phy ...OK

```
It seems that most things went fine, although a few issues. After

```
(si_env) matias@work:~/$ pip install herdingspikes
``` 
### herdingspikes

It looks like `herdingspikes` is now working on the check.

```
write_binary_recording with n_jobs 1  chunk_size None
Import spikeinterface ...OK
Import spikeinterface.full ...OK
Run tridesclous ...OK
RUNNING SHELL SCRIPT: spykingcircus_output/run_spykingcircus.sh
Run spykingcircus ...OK
# Generating new position and neighbor files from data file
# Not Masking any Channels
# Sampling rate: 30000
# Localization On
# Number of recorded channels: 16
# Few recording channels: not subtracing mean from activity
# Analysing frames: 6000000; Seconds: 200.0
# Frames before spike in cutout: 9
# Frames after spike in cutout: 54
# tcuts: 39 84
# tInc: 100000
# Detection completed, time taken: 0:00:02.426973
# Time per frame: 0:00:00.000404
# Time per sample: 0:00:00.000025
Loaded 3617 spikes.
Fitting dimensionality reduction using all spikes...
...projecting...
...done
Clustering...
Clustering 3617 spikes...
number of seeds: 47
seeds/job: 2
using 24 cpus
[... 119 lines cut ...]
def _clean():
    # clean
    folders = [
        'toy_example_recording',
        "tridesclous_output", "tridesclous_waveforms",
        "spykingcircus_output", "spykingcircus_waveforms",
        "phy_example"
    ]
    for folder in folders:
        if Path(folder).exists():
            shutil.rmtree(folder)
```
I would also appreciate if there was an option to avoid deleting all folders. Something quick and dirty like this might do:

```
keep_things = input("Do you want to keep installation check folders [y/n]? :> ")
if keep_things.lower() == 'yes'  or 'y':
    print("Keeping folders generated by installation check")
    return
else:
    # clean
    ...
``` 

I am happy to submit PR of any of this. Thank you for your work in this project!

samuelgarcia, 2022-01-28:
Hello.
Thanks for report.

About the installation pf **herdingspikes**.
Yes, i think that herdingspikes need to have numpy already installed.
So the first install in the yaml do not work.
You should ask to the dev of herdingspikes.

About **spikeinterface-gui** issue : if you want it to work properly, for now you should work from source for both spikeinterface and spikeinterface-gui. We plan a release soon. For this maybe you should also install pyside instead of PyQt5 but in that case phy won't work.

Generally this **installation_tips** folder is more for windows/osx users.
For ubuntu user, we recommand apt + pip to solve installation problem because it is what we do so it is easier to tests.
Maybe it could be usefull to have a  specific readme for that. A general bash script to install everything for ubuntu sounds difficult because the way users handle python in ubuntu is ultra diverse (conda vs pip + apt, vitualenv vs venv, ...).
So maybe if you want to make a PR you could make a **installation_in_ubuntu.md** file.
And maybe you could make this file in 2 parts : install from release and install from source.

alejoe91, 2022-08-01:
closed for inactivity