SpikeInterface/spikeinterface #484: `plot_timeseries` not showing traces

matiasandina, 2022-01-28:
I am experiencing some issues with `plot_timeseries`. Not sure if I'm missing something on my end.
First, I couldn't get the expected results from command line. I guess it's something related with `matplotlib` backend. If possible, it would be nice to have this command line compatible and not force the pipelines towards notebooks. 

When running on a `jupyter lab`, I could reproduce the example in the docs using the `mearec` dataset that I downloaded from the repository.

```
mport matplotlib.pyplot as plt
import spikeinterface as si  # import core only
import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw

local_path = 'mearec_test_10s.h5'
recording, sorting_true = se.read_mearec(local_path)
print(recording)
print(sorting_true)

w_ts = sw.plot_timeseries(recording, time_range=(0, 5))

```
[image]

I wanted to check the traces made sense before going into the sorting part with my own dataset. But when using my dataset I can't generate the traces:

```
import numpy as np
import matplotlib
import spikeinterface as si  # import core only
import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw

filename = "raw_uv_hpc_and_ms_local_ref2021-10-04T12_20_15.dat"
print(f"Reading {filename}")
recording = si.BinaryRecordingExtractor(filename, 
		sampling_frequency=30000.0,
		num_chan = 60,
		dtype=np.float32, 
		t_starts=None, 
		channel_ids=None, time_axis=0, 
		file_offset=0, gain_to_uV=None, 
		offset_to_uV=None, 
		is_filtered=None)

samp_freq = recording.get_sampling_frequency()
num_chan = recording.get_num_channels()

trace_snippet = recording.get_traces(start_frame=int(samp_freq * 0), end_frame=int(samp_freq * 2))
trace_snippet
print(f"trace_snippet shape is {trace_snippet.shape}")
w_ts = sw.plot_timeseries(recording, time_range=(0, 5))
```

No errors are found and the dataset is read into proper shape, but no plot is shown

```
Reading raw_uv_hpc_and_ms_local_ref2021-10-04T12_20_15.dat
trace_snippet shape is (60000, 60)
```
When I try to check the object, or force the plot, still nothing

```
w_ts
<spikeinterface.widgets.timeseries.TimeseriesWidget at 0x7ff1617d3e80>
w_ts.plot()
```

I tried to inspect the data in matrix form, and it seems the import is happening as expected

```
traces0 = recording.get_traces(segment_index=0, start_frame=int(samp_freq * 0), end_frame=int(samp_freq * 2))
traces0
memmap([[ -42.510254,  -71.56494 , -246.47998 , ...,  111.34473 ,
           28.080078,    0.      ],
        [-127.33496 , -172.37988 , -335.3999  , ...,   78.58496 ,
           48.359863,    0.      ],
        [-244.7251  , -311.80518 , -469.9502  , ...,   61.424805,
           43.680176,    0.      ],
        ...,
        [ -50.31006 ,  207.28516 ,  523.96484 , ...,   65.71484 ,
          -67.08008 ,    0.      ],
        [ -54.600098,  211.5752  ,  519.6748  , ...,   61.424805,
          -70.78516 ,    0.      ],
        [ -54.20996 ,  213.13525 ,  518.5049  , ...,   61.425293,
          -75.07471 ,    0.      ]], dtype=float32)
```

Data looks identical as what I can read "manually" in MATLAB or python. I have a dataset that is `np.int16` and it also doesn't show when plotted. I tried the following pre-processing steps in the pipeline and met no errors. I assume I would be able to go on with sorting and post-processing but I would prefer being able to see the raw input.

alejoe91, 2022-01-28:
Hi @matiasandina 

Could it be that the plot doesn't show because of the `memmap`? Can you plot directly `traces0` with matplotlib?

Cheers
Alessio

matiasandina, 2022-01-31:
Hello, I'm sorry the delay. After a reboot from my system for other reasons, I couldn't reproduce this error...Looks like there was something messing up the environment, but it's gone.

alejoe91, 2022-01-31:
Great to hear! Closing then :)

closed 2022-01-31 by alejoe91