SpikeInterface/spikeinterface #486: NeuroScopeRecordingExtractor fails to initialize when not every channel has a group in XML file

h-mayorquin, 2022-02-01:
Hi, this is most likely a `neo` issue. The following XML file comes from one of our conversions in CatalystNeuro.

https://www.dropbox.com/s/oyjq3qnh63fpngp/DT1_rLS_20150723_1584um.xml?dl=0

When the class is initialized with the data corresponding data of the XML file 
 the following error arises: 
 
[image]

I took a look at the code in Neo and basically the error comes from the assumption that every channel will have an assigned group (see the loop in the code):
https://github.com/NeuralEnsemble/python-neo/blob/2e00c994e46a1396fb4bdd393535604e6fc5568f/neo/rawio/neuroscoperawio.py#L74-L84

Which for the file above it is not true, here are the values of `channel_group`:
```python
{0: 0, 2: 0, 4: 0, 6: 0, 8: 1, 10: 1, 12: 1, 14: 1, 9: 2, 11: 2, 13: 2, 15: 2, 3: 3, 5: 3, 7: 3, 16: 4, 18: 4, 20: 4, 22: 4, 17: 5, 19: 5, 21: 5, 23: 5, 24: 6, 26: 6, 28: 6, 30: 6, 25: 7, 29: 7, 31: 7}
```
As you can see, the channel 1 is missing as a key in this dictionary (i.e. no group is assigned).

I can prepare a more complete example but first I wanted to bring it up as it might be a `Neo` issue and probably @samuelgarcia or @alejoe91 would have some advice on how to handle this.

alejoe91, 2022-02-01:
Thank you @h-mayorquin 

I think this should be an easy fix in NEO. Do you want to give it a try? Can you open an issue there too?

samuelgarcia, 2022-02-01:
Hi. Yes the fix fix sounds easy. Is this channel_group used for something else than the channel name in the code ?

samuelgarcia, 2022-02-01:
You should open an issue on neo.

h-mayorquin, 2022-02-03:
Yes, I can give it a try in Neo. I just wanted to double check that it made sense as an error. I will open an issue there and a PR there.

samuelgarcia, 2022-03-15:
No it is not solve already!! Sorry. I will do it.

closed 2022-04-01 by alejoe91