SpikeInterface/spikeinterface #500: Correction for differences in sampling time across channels

agbondy, 2022-02-04:
Neuropixels (and other MUAs) generally do not sample each channel at the same time due to bandwidth limitations. So nominally simultaneous samples actually reflect non-simultaneous events. The exact offset between sampling times of each channel is known. For Neuropixels probes acquired through spikeGLX, there is an existing tool ([catgt](https://billkarsh.github.io/SpikeGLX/help/catgt_tshift/catgt_tshift/)) to resample the data recorded on each channel at exactly simultaneous times. This makes for much more accurate common average referencing, and is particularly useful for removing fast transients where accuracy in timing is critical.

Does SpikeInterface offer such a capability? If not, this may be worth considering as a future improvement.

alejoe91, 2022-02-04:
Hi @agbondy 

We are aware of this and porting the IBL code to remove such offsets in on our (long) TODO list :)
My estimate is that we could get it done by the end of february.

alejoe91, 2022-04-01:
WIP https://github.com/SpikeInterface/spikeinterface/pull/508

closed 2022-08-01 by alejoe91