SpikeInterface/spikeinterface #510: Missing Probeinterface file for Neuropixel 1.0 (Open Ephys)

phanhuynh, 2022-02-14:
Hello everyone!  

I'm using spikeinterface to analyze Neuropixel (1.0) data from Open Ephys.    
However, I'm missing the ProbeInterface probe object.  

I've tried looking in ProbeInterface public library.  And I'd like to avoid generating it from scratch.  Does anyone know where I can find it?    

Thank you!  I'm new to this so your support is deeply appreciated.

samuelgarcia, 2022-02-14:
I don't known how the OE project manage the probe channel mapping for neuropixel.
Is there any file in the OE folder we could rely on to make the probe generation automatic ?

alejoe91, 2022-02-21:
Discussed and fixed here: https://github.com/SpikeInterface/probeinterface/issues/85

closed 2022-02-21 by alejoe91