SpikeInterface/spikeinterface #513: Issue with PhySortingExtractor

vincisLab, 2022-02-17:
Hi,

we are trying to load back unit information after manual curation in phy.
When we input: 
sorting_phy_curated = se.PhySortingExtractor('D:/DataTemp/CB287/021622/Sorting/all_sorters/0/ironclust/phy_TS/', exclude_cluster_groups=['noise'])_
we get the following error.

Any suggestion on the best way to solve it?
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
D:\AnacondaStuff\envs\si_env\lib\site-packages\pandas\core\indexes\base.py in get_loc(self, key, method, tolerance)
   3360             try:
-> 3361                 return self._engine.get_loc(casted_key)
   3362             except KeyError as err:

D:\AnacondaStuff\envs\si_env\lib\site-packages\pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_loc()

D:\AnacondaStuff\envs\si_env\lib\site-packages\pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'id'

The above exception was the direct cause of the following exception:

KeyError                                  Traceback (most recent call last)
C:\Users\VINCIS~1\AppData\Local\Temp/ipykernel_7728/3039858487.py in <module>
----> 1 sorting_IC_phy_curated = se.PhySortingExtractor('D:/DataTemp/CB287/021622/Sorting/all_sorters/0/ironclust/phy_TS/', exclude_cluster_groups=['noise'])

D:\AnacondaStuff\envs\si_env\lib\site-packages\spikeinterface\extractors\phykilosortextractors.py in __init__(self, folder_path, exclude_cluster_groups)
    134 
    135     def __init__(self, folder_path, exclude_cluster_groups=None):
--> 136         BasePhyKilosortSortingExtractor.__init__(self, folder_path, exclude_cluster_groups, keep_good_only=False)
    137 
    138         self._kwargs = {'folder_path': str(Path(folder_path).absolute()),

D:\AnacondaStuff\envs\si_env\lib\site-packages\spikeinterface\extractors\phykilosortextractors.py in __init__(self, folder_path, exclude_cluster_groups, keep_good_only)
     91         if keep_good_only and "KSLabel" in cluster_info.columns:
     92             cluster_info = cluster_info.query(f"KSLabel != 'good'")
---> 93         unit_ids = cluster_info["id"].values
     94 
     95         BaseSorting.__init__(self, sampling_frequency, unit_ids)

D:\AnacondaStuff\envs\si_env\lib\site-packages\pandas\core\frame.py in __getitem__(self, key)
   3456             if self.columns.nlevels > 1:
   3457                 return self._getitem_multilevel(key)
-> 3458             indexer = self.columns.get_loc(key)
   3459             if is_integer(indexer):
   3460                 indexer = [indexer]

D:\AnacondaStuff\envs\si_env\lib\site-packages\pandas\core\indexes\base.py in get_loc(self, key, method, tolerance)
   3361                 return self._engine.get_loc(casted_key)
   3362             except KeyError as err:
-> 3363                 raise KeyError(key) from err
   3364 
   3365         if is_scalar(key) and isna(key) and not self.hasnans:

KeyError: 'id'

samuelgarcia, 2022-02-18:
Hi.
Thanks for the report.
@alejoe91 : this is for you!!

alejoe91, 2022-02-18:
I think this is fixed in the latest release (https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/extractors/phykilosortextractors.py#L89-L95).

Can you try to `pip install --upgrade spikeinterface` and run it again??

vincisLab, 2022-02-18:
Hi Alessio,

After running pip install --upgrade spikeinterface, it is still giving us the same error.

alejoe91, 2022-02-18:
Can you paste the new error so I can see the correct lines? Note that you can use the `code` formatting (<>) for better readibility

vincisLab, 2022-02-18:
We are trying to reinstall spikeinterface from source and see if that solve the issue.
I will keep you posted.

alejoe91, 2022-02-18:
Would you mind sending me all the .tsv files in the phy/kilosort folder?

vincisLab, 2022-02-18:
[cluster_si_unit_ids.zip](https://github.com/SpikeInterface/spikeinterface/files/8099916/cluster_si_unit_ids.zip)

Here you go.

Just as an update we reinstall spikeinterface from source. Everything looks good except now we have problems launching Phy ...We will figure it out eventually. I attached the file in a zip folder.

vincisLab, 2022-02-20:
Ok, we solved the issue with Phy.
Yes, the problem of opening the phy folder is solved with master.

Thanks!

closed 2022-02-20 by alejoe91