SpikeInterface/spikeinterface #522: Another issue with PhySortingExtractor

vincisLab, 2022-02-22:
Hi,

we enocunter anothe issue with PhySortingExtractor.
Once we run the extractor after curation in Phy we noticed the following:

"get_unit_ids" in spikeinterface is giving us the  index of the units shown in "si_unit_id" of the ClusterView (Phy GUI).
However, the spike trains information are stored under the index "id" of the ClusterView (Phy GUI). 
Since the "id" and the "si_unit_id" of the ClusterView (Phy GUI) often do not match (especially after merging or splitting during manual curation), this creates issues when extracting the timestamps of the spikes (and even the waveforms) when using "get_unit_ids" to iterate through the id of the unit after manula curation.

I hope this is clear, otherwise I am happy to give more details.

vincisLab, 2022-02-23:
Hi @alejoe91 

It is correct. 
However,  sometimes the probelm arises even  even without merging or splitting in Phy because the  "id" and the "si_unit_id" of the ClusterView do not match and PhySortingExtractor use the si_unit_id.
For example I am looking now at Phy_TS if a unit has a Phy "id"=1 and a Phy "si_unit_id"=12,  PhySortingExtractor is storing the spiketimes with index 1 but the .get_unit_ids will give you 12.

alejoe91, 2022-02-23:
Thanks for the clarification. Do you mind sharing with me a couple of zipped phy folders so I can take a closer look? You can exclude the .dat file ;)

vincisLab, 2022-02-23:
@alejoe91  here a link https://www.dropbox.com/sh/1mmy91g5ypcl3ug/AAAZnbK7bdKuG2ZF1pEsnm5Qa?dl=0

There are two phy folders.

Roberto

alejoe91, 2022-02-23:
Ciao Roberto,

Can you try this fix? https://github.com/SpikeInterface/spikeinterface/pull/525

Basically I'm mapping the cluster ids from the phy to the SI ids. Sorry we didn't catch it before and thank you for reporting the issue :)

vincisLab, 2022-02-24:
Ciao Alessio,

yes, it indeed solved the issue...or at least partially.
We do have a case of phy folder in which there was substantial manual curation (merging and splitting). In that example, the sorting extractor still fail to report the correct ids and instead give us a bunch of NaNs.
If you want i can share the Phy folder with you. Let me know.

alejoe91, 2022-02-24:
Yes that would be great! I think that in that case we might need to create some "new" SI ids because Phy doesn't do it automatically. I guess that the si_unit_id column in that case becomes blank..right?

vincisLab, 2022-02-24:
Yes you are right it is blank.
I am adding the zip file (Example_3) in the same shared dropbox folder (link above in our previous messages).

alejoe91, 2022-02-25:
@vincisLab I pushed a fix that deals with nan si_unit_ids in the same PR. Let us know if that solves all issues :)

vincisLab, 2022-02-25:
Perfect! It works great now!

Thanks!

closed 2022-02-25 by alejoe91