SpikeInterface/spikeinterface #537: [doc] bad default documentation version

JuliaSprenger, 2022-03-04:
Currently https://spikeinterface.readthedocs.io redirects to https://spikeinterface.readthedocs.io/en/latest/ which does not exist. The default version should probably be https://spikeinterface.readthedocs.io/en/stable/ instead.

samuelgarcia, 2022-03-04:
Hi Julia,
the activity on spikeinterface is so high that most of current users install from sources.
So having by default "latest" instead of "stable" is a good choice at the moment, I think.
After 1.0.0 release we will change this of course :)
@alejoe91 are you agree ?

JuliaSprenger, 2022-03-04:
Hi @samuelgarcia. The problem is that `latest` points to a page telling `SORRY  - This page does not exist yet.` Could you then fix the doc compilation for the latest version?

samuelgarcia, 2022-03-04:
Oh toally sorry. I did not understand. I will check.

alejoe91, 2022-03-04:
The doc compilation succeeded..very strange, it was working until yesterday! Maybe something is down with RTD?

joramvanrheede, 2022-03-07:
Still having the same issue today. For me / currently, it's also not able to open _https://spikeinterface.readthedocs.io/en/stable/_ 
giving the same _Maze Found - SORRY This page does not exist yet_ error.

siddjakes, 2022-03-09:
It seems to be loading fine now. Thank you all!

JuliaSprenger, 2022-03-09:
I can confirm `latest` works now for be, but `stable` remains broken.

samuelgarcia, 2022-03-09:
This is strange because 0.93 url work and stable is supposed to be 0.93!

closed 2022-04-01 by alejoe91