SpikeInterface/spikeinterface #538: export_to_phy error: Input contains NaN, infinity or a value too large for dtype('float32')

phanhuynh, 2022-03-04:
Hello folks! I'm using kilosort3 on neuropixel 1.0 data in spikeinterface: This error occurs when I export waveforms to phy:
Thank you!
```
Warning: empty units have been removed when being exported to Phy
write_binary_recording with n_jobs 1  chunk_size 3125000
write_binary_recording: 100%|############################################################| 2/2 [00:00<00:00, 17.85it/s]
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [29], in <module>
      1 start_time = time.time()
----> 2 export_to_phy(we_KS3, output_folder='_phy_KS3',
      3               progress_bar=True, total_memory='100M')
      4 print("--- _phy_KS3 took %s seconds ---" % (time.time() - start_time))

File ~\anaconda3\envs\siphy_env\lib\site-packages\spikeinterface\exporters\to_phy.py:161, in export_to_phy(waveform_extractor, output_folder, compute_pc_features, compute_amplitudes, sparsity_dict, copy_binary, max_channels_per_template, remove_if_exists, peak_sign, template_mode, dtype, verbose, **job_kwargs)
    158     templates.append(template_full)
    159     templates_ind.append(inds_full)
--> 161 template_similarity = compute_template_similarity(waveform_extractor, method='cosine_similarity')
    163 np.save(str(output_folder / 'templates.npy'), templates)
    164 np.save(str(output_folder / 'template_ind.npy'), templates_ind)

File ~\anaconda3\envs\siphy_env\lib\site-packages\spikeinterface\toolkit\postprocessing\template_similarity.py:37, in compute_template_similarity(waveform_extractor, waveform_extractor_other, method)
     35     else:
     36         templates_other_flat = None
---> 37     similarity = sklearn.metrics.pairwise.cosine_similarity(templates_flat, templates_other_flat)
     38 # elif method == '':
     39 else:
     40     raise ValueError(f'compute_template_similarity(method {method}) not exists')

File ~\anaconda3\envs\siphy_env\lib\site-packages\sklearn\metrics\pairwise.py:1251, in cosine_similarity(X, Y, dense_output)
   1217 """Compute cosine similarity between samples in X and Y.
   1218 
   1219 Cosine similarity, or the cosine kernel, computes similarity as the
   (...)
   1247 kernel matrix : ndarray of shape (n_samples_X, n_samples_Y)
   1248 """
   1249 # to avoid recursive import
-> 1251 X, Y = check_pairwise_arrays(X, Y)
   1253 X_normalized = normalize(X, copy=True)
   1254 if X is Y:

File ~\anaconda3\envs\siphy_env\lib\site-packages\sklearn\metrics\pairwise.py:147, in check_pairwise_arrays(X, Y, precomputed, dtype, accept_sparse, force_all_finite, copy)
    144     dtype = dtype_float
    146 if Y is X or Y is None:
--> 147     X = Y = check_array(
    148         X,
    149         accept_sparse=accept_sparse,
    150         dtype=dtype,
    151         copy=copy,
    152         force_all_finite=force_all_finite,
    153         estimator=estimator,
    154     )
    155 else:
    156     X = check_array(
    157         X,
    158         accept_sparse=accept_sparse,
   (...)
    162         estimator=estimator,
    163     )

File ~\anaconda3\envs\siphy_env\lib\site-packages\sklearn\utils\validation.py:800, in check_array(array, accept_sparse, accept_large_sparse, dtype, order, copy, force_all_finite, ensure_2d, allow_nd, ensure_min_samples, ensure_min_features, estimator)
    794         raise ValueError(
    795             "Found array with dim %d. %s expected <= 2."
    796             % (array.ndim, estimator_name)
    797         )
    799     if force_all_finite:
--> 800         _assert_all_finite(array, allow_nan=force_all_finite == "allow-nan")
    802 if ensure_min_samples > 0:
    803     n_samples = _num_samples(array)

File ~\anaconda3\envs\siphy_env\lib\site-packages\sklearn\utils\validation.py:114, in _assert_all_finite(X, allow_nan, msg_dtype)
    107     if (
    108         allow_nan
    109         and np.isinf(X).any()
    110         or not allow_nan
    111         and not np.isfinite(X).all()
    112     ):
    113         type_err = "infinity" if allow_nan else "NaN, infinity"
--> 114         raise ValueError(
    115             msg_err.format(
    116                 type_err, msg_dtype if msg_dtype is not None else X.dtype
    117             )
    118         )
    119 # for object dtype data, we only check for NaNs (GH-13254)
    120 elif X.dtype == np.dtype("object") and not allow_nan:

ValueError: Input contains NaN, infinity or a value too large for dtype('float32').
```

alejoe91, 2022-03-04:
Hey @phanhuynh 

Seems like there are some issues with waveforms! Can you extract waveforms and check the templates. manually?

samuelgarcia, 2022-03-04:
Hi,
@alejoe91 this is strange because templates that have problems (like no spikes) are not set to nan but to zeros.
So similarioyty should work no ?

phanhuynh, 2022-03-04:
Hello @samuelgarcia , what should I check in the templates?  I've zipped the waveforms in case it helps.
Thank you soooo much!
[_waveform_KS3.zip](https://github.com/SpikeInterface/spikeinterface/files/8185653/_waveform_KS3.zip)

samuelgarcia, 2022-03-04:
Thanks.
@alejoe91 : in this zip file "templates_average.npy" have some nan.
how is it possible ? we have to check this.

alejoe91, 2022-03-04:
We have to check templates IMO @samuelgarcia 

Maybe they have nan values..which would be strange

alejoe91, 2022-03-04:
Ok so the problem is that you have a unit with 0 spikes!

In order to avoid this, you can do:
```
unit_with_spikes = []
for unit in sorting.get_unit_ids():
    if len(sorting.get_unit_spike_train(unit)) > 0:
        unit_with_spikes.append(unit)
sorting_with_spikes = sorting.select_units(units_with_spikes)
```

And you can then proceed to export to phy with the `sorting_with_spikes` objects. @samuelgarcia we should add an automated function that does this

[image]

alejoe91, 2022-03-04:
https://github.com/SpikeInterface/spikeinterface/pull/540

samuelgarcia, 2022-03-04:
@alejoe91 : maybe for a zero spikes units we could have a template with zero instead  nan, no ?

phanhuynh, 2022-03-07:
Again, thank you so much for all your effort to the community!

phanhuynh, 2022-03-07:
It worked perfectly!  Again, thank you guys so much for your efforts to this community!  Amazing :)

alejoe91, 2022-03-07:
Closing then!

closed 2022-03-07 by alejoe91