SpikeInterface/spikeinterface #541: How to merge sortings after extraction?

valejandracch, 2022-03-07:
Hello, i have a question.
I have some Klusta  sortings and I want to load the spike sorting output from the spike sorting folder to do some postprosessing and automatic curation with them, but i see that in the tutorial you say that:
 you can reload the output as a SortingExtractor using the corresponding spikeextractors class. Note that if sorting by group/property, single groups must be loaded separately:
sorting_KL_0 = se.KlustaSortingExtractor("results_split_klusta/0")

I have separetad groups indeed, and i would like to know if there is some way to merge that sorting objects into one again instead of one for each group.

Im using the version 13. thanks!

ferchaure, 2022-03-08:
I use the class [UnitsAggregationSorting](https://github.com/SpikeInterface/spikeinterface/blob/04068c48eb590a1f6cfd7db1b4aef47d8d37fb2f/spikeinterface/core/unitsaggregationsorting.py) imported from spikeinterface, giving a list with the sortings as the input.

alejoe91, 2022-03-08:
Thanks for the reply @ferchaure 

Indeed the `UnitsAggregationSorting` class is the approach for newer versions (>=0.90).

For the old 0.13 version, you can use the [`MultiSortingExtractor`](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/multisortingextractor.py), which does exactly the same

valejandracch, 2022-03-08:
Thanks @alejoe91 and @ferchaure  for the help!!
Actually i just found the MultiSortingExtractor this morning and i give it a try like this:
 It worked.

`import os
os.listdir(output_folderKL)

slist = [se.KlustaSortingExtractor(output_folderKL+'/'+sF) for sF in os.listdir(output_folderKL)]
n=0
for KLs in slist:
    units=KLs.get_unit_ids()
    groups= KLs.get_units_property(unit_ids= units, property_name='group')
    groups=list(map(lambda x : x + n, groups))
    KLs.set_units_property(unit_ids=units,property_name='group',values=groups)
    print(f'Klusta unit ids group: {units}')
    print(f'Klusta unit ids group: {groups}')
    n=n+1

sorting_KL=se.MultiSortingExtractor(slist)
units=sorting_KL.get_unit_ids()
groups= sorting_KL.get_units_property(unit_ids= units, property_name='group')
print(f'Klusta unit ids group: {units}')
print(f'Klusta unit ids group: {groups}')`

alejoe91, 2022-03-08:
Great! Closing the issue then!

closed 2022-03-08 by alejoe91