SpikeInterface/spikeinterface #547: Extracting Waveform  from an Agreement sorting

Kayv-cmb, 2022-03-10:
Hello,
I'm trying to run a consensus base spikesorting method, I compare multiple sorter using compare_multiple_sorters then i get the agreement by agr_3 = mcmp.get_agreement_sorting(minimum_agreement_count=3). However I want to export to phy the result of my agreement, but i can't extract the waveform of my agreement and i get this error 
AttributeError: 'MultiSortingComparison' object has no attribute '_spiketrains'
When i use the function extract_waveform.
I tried to save it before exporting it but i got the same error, 
So my question is it possible to export to phy an agreement sorting?

Here is my code

**mcmp = sc.compare_multiple_sorters([sorting_KS, sorting_IC, sorting_MS, sorting_TDC_5], ['KS', 'IC', 'MS4', 'TDC'], 
                                       spiketrain_mode='union', verbose=True)
w = sw.plot_multicomp_agreement(mcmp)
w = sw.plot_multicomp_agreement_by_sorter(mcmp)
sw.plot_multicomp_graph(mcmp)
agr_3 = mcmp.get_agreement_sorting(minimum_agreement_count=3)

we_ms4 = si.WaveformExtractor.create(recording_4_tetrodes, agr_3, 'waveforms', remove_if_exists=True)
we_ms4.set_params(ms_before=3., ms_after=4., max_spikes_per_unit=500)
we_ms4.run_extract_waveforms(n_jobs=-1, chunk_size=30000)
print(we_ms4)
from spikeinterface.exporters import export_to_phy
export_to_phy(recording_4_tetrodes, we_ms4 , 'phy_AGR', verbose=False)**

samuelgarcia, 2022-03-10:
Hi,
could you show the full trace please ?
maybe save sorting to a foldert can help
```python
agr_3 = agr_3.save(folder='/path/to/a/folder')
```

Kayv-cmb, 2022-03-10:
I just tried to save to a folder did not work either

Here is the full traceback : 
mcmp = sc.compare_multiple_sorters([sorting_KS, sorting_IC, sorting_MS, sorting_TDC_5], ['KS', 'IC', 'MS4', 'TDC'], 
                                       spiketrain_mode='union', verbose=True)
w = sw.plot_multicomp_agreement(mcmp)
w = sw.plot_multicomp_agreement_by_sorter(mcmp)
sw.plot_multicomp_graph(mcmp)
agr_3 = mcmp.get_agreement_sorting(minimum_agreement_count=2)
agr_3 = agr_3.save(folder='D:/spiketutorials-master/spikesortingTetrode/Mallory/AGR_saved')
we_ms4 = si.WaveformExtractor.create(recording_4_tetrodes, agr_3, 'waveforms', remove_if_exists=True)
we_ms4.set_params(ms_before=3., ms_after=4., max_spikes_per_unit=500)
we_ms4.run_extract_waveforms(n_jobs=-1, chunk_size=30000)
print(we_ms4)
from spikeinterface.exporters import export_to_phy
export_to_phy(recording_4_tetrodes, agr_3, 'phy_AGR', verbose=False)
Multicomaprison step 1: pairwise comparison
  Comparing: KS and IC
  Comparing: KS and MS4
  Comparing: KS and TDC
  Comparing: IC and MS4
  Comparing: IC and TDC
  Comparing: MS4 and TDC
Multicomparison step 2: make graph
Multicomaprison step 3: clean graph
Removed 0 duplicate nodes
Multicomparison step 4: extract agreement from graph
C:\Users\kayva\.conda\envs\environment\lib\site-packages\networkx\drawing\nx_pylab.py:717: MatplotlibDeprecationWarning: Passing *transOffset* without *offsets* has no effect. This behavior is deprecated since 3.5 and in 3.6, *transOffset* will begin having an effect regardless of *offsets*. In the meantime, if you wish to set *transOffset*, call collection.set_offset_transform(transOffset) explicitly.
  edge_collection = mpl.collections.LineCollection(
Traceback (most recent call last):

  File "C:\Users\kayva\AppData\Local\Temp/ipykernel_17176/3394243740.py", line 7, in <module>
    agr_3 = agr_3.save(folder='D:/spiketutorials-master/spikesortingTetrode/Mallory/AGR_saved')

  File "C:\Users\kayva\.conda\envs\environment\lib\site-packages\spikeinterface\core\base.py", line 579, in save
    loaded_extractor = self.save_to_folder(**kwargs)

  File "C:\Users\kayva\.conda\envs\environment\lib\site-packages\spikeinterface\core\base.py", line 654, in save_to_folder
    cached = self._save(folder=folder, verbose=verbose, **save_kwargs)

  File "C:\Users\kayva\.conda\envs\environment\lib\site-packages\spikeinterface\core\basesorting.py", line 115, in _save
    NpzSortingExtractor.write_sorting(self, save_path)

  File "C:\Users\kayva\.conda\envs\environment\lib\site-packages\spikeinterface\core\npzsortingextractor.py", line 54, in write_sorting
    sp_ind = sorting.get_unit_spike_train(unit_id, segment_index=seg_index)

  File "C:\Users\kayva\.conda\envs\environment\lib\site-packages\spikeinterface\core\basesorting.py", line 60, in get_unit_spike_train
    spike_train = segment.get_unit_spike_train(

  File "C:\Users\kayva\.conda\envs\environment\lib\site-packages\spikeinterface\comparison\multicomparisons.py", line 234, in get_unit_spike_train
    spiketrain = np.array(self._msc._spiketrains[ind])

AttributeError: 'MultiSortingComparison' object has no attribute '_spiketrains'

samuelgarcia, 2022-03-10:
OK I will check.

closed 2022-03-15 by samuelgarcia