SpikeInterface/spikeinterface #549: Unable to extract whitened waveforms

lfrank, 2022-03-14:
We calculate metrics on sortings using whitened waveforms, but currently this seems not to be supported, as when I try to whiten the recording extractor and then get the waveforms, I get an error. Any thoughts would be appreciated.

The code snippet is 

`       if waveform_params['whiten']:
            recording = toolkit.preprocessing.whiten(recording)
        waveforms = si.extract_waveforms(recording=recording, 
                                         sorting=sorting, 
                                         folder=key['waveform_extractor_path'],
                                         **waveform_params)`

Error (apologies for the long path information strings):

`[File ~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py:62, in Waveforms.make(self, key)
     ]()[60](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=59)[ if waveform_params['whiten']:
     ]()[61](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=60)[     recording = toolkit.preprocessing.whiten(recording)
---> ]()[62](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=61)[ waveforms = si.extract_waveforms(recording=recording, 
     ]()[63](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=62)[                                  sorting=sorting, 
     ]()[64](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=63)[                                  folder=key['waveform_extractor_path'],
     ]()[65](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=64)[                                  **waveform_params)
     ]()[67](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=66)[ key['analysis_file_name'] = AnalysisNwbfile().create(key['nwb_file_name'])
     ]()[68](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=67)[ object_id = AnalysisNwbfile().add_units_waveforms(key['analysis_file_name'],
     ]()[69](file:///~/Src/NWB/nwb_datajoint/src/nwb_datajoint/common/common_waveforms.py?line=68)[                                                   waveform_extractor=waveforms)

File ~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py:798, in extract_waveforms(recording, sorting, folder, load_if_exists, precompute_template, ms_before, ms_after, max_spikes_per_unit, unselect_spike_on_border, overwrite, return_scaled, dtype, use_relative_path, **job_kwargs)
    ]()[795](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=794)[ we = WaveformExtractor.create(recording, sorting, folder, use_relative_path=use_relative_path)
    ]()[796](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=795)[ we.set_params(ms_before=ms_before, ms_after=ms_after, max_spikes_per_unit=max_spikes_per_unit, dtype=dtype,
    ]()[797](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=796)[               return_scaled=return_scaled)
--> ]()[798](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=797)[ we.run_extract_waveforms(**job_kwargs)
    ]()[800](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=799)[ if precompute_template is not None:
    ]()[801](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=800)[     we.precompute_templates(modes=precompute_template)

File ~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py:682, in WaveformExtractor.run_extract_waveforms(self, **job_kwargs)
    ]()[679](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=678)[ spikes = np.concatenate(spikes)
    ]()[681](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=680)[ wf_folder = self.folder / 'waveforms'
--> ]()[682](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=681)[ wfs_arrays = extract_waveforms_to_buffers(self.recording, spikes, unit_ids, nbefore, nafter,
    ]()[683](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=682)[                         mode='memmap', return_scaled=return_scaled, folder=wf_folder, dtype=p['dtype'],
    ]()[684](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py?line=683)[                         sparsity_mask=None,  copy=False, **job_kwargs)

File ~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py:80, in extract_waveforms_to_buffers(recording, spikes, unit_ids, nbefore, nafter, mode, return_scaled, folder, dtype, sparsity_mask, copy, **job_kwargs)
     ]()[74](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=73)[ dtype = np.dtype(dtype)
     ]()[77](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=76)[ wfs_arrays, wfs_arrays_info = allocate_waveforms_buffers(recording, spikes, unit_ids, nbefore, nafter,
     ]()[78](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=77)[                     mode=mode, folder=folder, dtype=dtype, sparsity_mask=sparsity_mask)
---> ]()[80](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=79)[ distribute_waveforms_to_buffers(recording, spikes, unit_ids, wfs_arrays_info, nbefore, nafter, return_scaled,
     ]()[81](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=80)[                                 mode=mode, sparsity_mask=sparsity_mask, **job_kwargs)
     ]()[83](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=82)[ if mode == 'memmap':
     ]()[84](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=83)[     return wfs_arrays

File ~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py:229, in distribute_waveforms_to_buffers(recording, spikes, unit_ids, wfs_arrays_info, nbefore, nafter, return_scaled, mode, sparsity_mask, **job_kwargs)
    ]()[226](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=225)[     init_args = (recording.to_dict(), )
    ]()[227](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=226)[ init_args = init_args + (unit_ids, spikes, wfs_arrays_info, nbefore,
    ]()[228](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=227)[                          nafter, return_scaled, inds_by_unit, mode, sparsity_mask)
--> ]()[229](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=228)[ processor = ChunkRecordingExecutor(recording, func, init_func, init_args, job_name=f'extract waveforms {mode}',
    ]()[230](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=229)[                                    **job_kwargs)
    ]()[231](file:///~/opt/anaconda3/envs/nwb_datajoint/lib/python3.8/site-packages/spikeinterface/core/waveform_tools.py?line=230)[ processor.run()

TypeError: __init__() got an unexpected keyword argument 'whiten']()`

alejoe91, 2022-03-14:
Hi Loren! The 'problem' is that you are passing the `whiten` arg to the `extract_waveforms`, which is not supported. You can just add `del waveform_params['whiten']` before calling the waveform extraction function and that should fix it :)

lfrank, 2022-03-14:
> This Message Is From an External Sender
> This message came from outside your organization.

Oh, of course. Thanks for pointing that out.