SpikeInterface/spikeinterface #556: removing channels

vincisLab, 2022-03-17:
Hi all,

is there a way to remove some noisy channel/s from the recording after it has been linked to a specific probe ("probe") design [recording.set_probe(probe)], such that both the data (trace) as well as the info (position, plane axes, contact, etc..) of the electrode on probe will be deleted?

samuelgarcia, 2022-03-17:
Hi,
when you slect manually channels like this:
```python
some_channnel_ids = rec.channel_ids[0:10]
rec2 = rec.channel_slice(some_channnel_ids)
probe2 = rec2.get_probe()
plot_probe(probe2)
```
then the probe is also "sliced" along side so manual selection of good channel is easy.

If you want to automatically remove noisy channel you can use `remove_bad_channels()` from toolkit which is a trehold based version of this channel manual selection.

vincisLab, 2022-03-19:
Got it, thanks!!

closed 2022-03-25 by alejoe91