SpikeInterface/spikeinterface #559: Convert to uV - Gain and offset

mfvd, 2022-03-18:
Hello,

I'm now moving to the most recent version of spikeinterface and I'm not sure how to deal with unscaled signals.
In trying to convert to uV I tried the following code: 
```
rec.set_channel_offsets(offset)
rec.set_channel_gains(gain)
```
However, I'm not 100% if this matches the method used previously.

In the previous version (0.11.0) I scaled the signals when extracting the data form my binary file:

```
rec = se.BinDatRecordingExtractor(file,
                                  numchan=numchan,
                                  sampling_frequency=samp_freq,
                                  dtype=dtype,
                                  gain=0.195,
                                  offset=32768,
                                  gain_first=False)
```

Here using `gain_first=False` to first subtract the offest and multiply by the gain later (per intan documentation). Is the newest version achieving the same result? If not, how can I achieve a similar result?

Cheers

samuelgarcia, 2022-03-18:
In the new version, the signature change a bit, now you need to do
```python
rec = si.read_binary(file_paths, sampling_frequency, num_chan, dtype, t_starts=None, channel_ids=None,
                 time_axis=0, file_offset=0, gain_to_uV=None, offset_to_uV=None,
                 is_filtered=None)
```

`gain_first` disappear so you need to do  `offset_to_uV = 32768 * 0.195`

samuelgarcia, 2022-03-18:
Note that file_paths can be a list (multi segment) or a unique path (mono segment as before)

mfvd, 2022-03-21:
Thank you for your help! 
Cheers

samuelgarcia, 2022-03-21:
Santé!

closed 2022-03-21 by samuelgarcia