SpikeInterface/spikeinterface #560: Does `OldToNewRecordign` fail for all spikeinterface objects? Maybe better exception?

h-mayorquin, 2022-03-19:
Consider the following example:
```python
from spikeinterface.core.testing_tools import generate_recording
from spikeinterface.core.old_api_utils import OldToNewRecording

rec = generate_recording(num_channels=4, durations=[10])
OldToNewRecording(oldapi_recording_extractor=rec)
```

Produces the following trace:
```python
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/heberto/development/spikeinterface/spikeinterface/core/old_api_utils.py", line 175, in __init__
    oldapi_recording_extractor.get_dtype(return_scaled=False))
TypeError: get_dtype() got an unexpected keyword argument 'return_scaled'
``` 
My feeling is that this will consistently fail for any `spikeinterface` object, is this correct?
If so, maybe we should have a better exception handling for this one?

alejoe91, 2022-03-19:
The `generate_recording` already returns a new extractor! So the old_to_new is supposed to fail :)

h-mayorquin, 2022-03-19:
Yes, I believe that's what I should do. I was:
1) Confirming that it is indeed the case that I will consistently fail for any new extractor.
2) Proposing to add a more meaningful error that indicates that it should be a `spikeextractor` object.

h-mayorquin, 2022-03-19:
@alejoe91 I can take care of this (adding a more meaningful error) if you think is a good idea.

samuelgarcia, 2022-03-21:
Hi Herberto,
the complicated stuff is that we con't want that spikeinterface depend on spikeextractor.
So we can't afford a `import spikeextractors` in spikeinterface, so for testing if it is a spikeextractor object we have to check other stuff (class name, method names, ...).
Same for testing.

h-mayorquin, 2022-03-21:
Right, that makes sense.
But, can't we just check if `oldapi_recording_extractor` is "BaseExtractor" and  if it is return an error saying that this is intending for spikeextractor objects, etc.

I honestly don't think this is very important. I was just checking that the function did throw an error (which it does) but then I saw that the error trace was somehow obscure and I was wondering if it was in your interest to have a more explicit error.

closed 2022-08-01 by alejoe91