SpikeInterface/spikeinterface #561: [contribution] Windows-compatible container for Spike Interface

vncntprvst, 2022-03-19:
I heard something similar is in the works, but just in case this is of interest to anyone in the meantime, I made a Jupyterlab container for Spike Interface with a few sorters pre-installed. It can run on any OS, but could make life easier for Windows users. 
The repo is here: https://github.com/vncntprvst/spikeinterface-dockerfiles. 
Any feedback welcome! 
Thanks

alejoe91, 2022-04-01:
Thank you so much @vncntprvst 

Can you make  a PR with the Dockerfile here: https://github.com/SpikeInterface/spikeinterface-dockerfiles/pulls ?

So we can push it to our dockerhub too!

vncntprvst, 2022-04-01:
Sure. I'll make a new branch on my side for the PR, to isolate the relevant commits.

closed 2022-05-26 by vncntprvst