SpikeInterface/spikeinterface #565: Bug: feeding small batches to incremental PCA

miketrumpis, 2022-03-22:
I just ran into an error while exporting to phy. The issue came up in these lines, where the number of waveforms was nonzero, but still less than the smallest batch size

https://github.com/SpikeInterface/spikeinterface/blob/31911008a6dd8acebc09c8dce322d1a2118d2266/spikeinterface/toolkit/postprocessing/principal_component.py#L355-L361

I've monkey-patched this to this, and it appears to work. Might want to buffer that inequality with some margin for robustness.

```python
    for unit_id in unit_ids:
        wfs = we.get_waveforms(unit_id)
        if len(wfs) < p['n_components']:
            continue
```

While not a show-stopper, I get worrisome "unclosed file" ResourceWarnings in this same class in relation to this line: https://github.com/SpikeInterface/spikeinterface/blob/31911008a6dd8acebc09c8dce322d1a2118d2266/spikeinterface/toolkit/postprocessing/principal_component.py#L118 

Warnings go away using a `with open` pattern.

samuelgarcia, 2022-03-23:
Hi,
good catch. Could you make a PR for this ?

miketrumpis, 2022-03-24:
here you go #571

alejoe91, 2022-03-25:
fixed via https://github.com/SpikeInterface/spikeinterface/pull/571

closed 2022-03-25 by alejoe91