SpikeInterface/spikeinterface #572: Is there a way to sort by group using only one sorter?

mfvd, 2022-03-25:
Hello,

According to the documentation, it is possible to split by group and then use `run_sorters` to get a sorter object per group. However, if possible, I would like to a single object with data from all my groups (shanks)

In version 0.11.0 one could use `grouping_property='group'` to sort by group and to keep the data in a single sorter object. Is there a similar method with 0.93.0? 

Cheers

alejoe91, 2022-03-25:
HI @mfvd 

Sure! Checkout the [run_sorter_by_property()](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/launcher.py#L46) function! I think it's exactly what you need :)

mfvd, 2022-03-25:
Hi @alejoe91 ,

Thanks! 
I completely missed that one when reading the docs. I'll give it a try!

Cheers

alejoe91, 2022-03-25:
Ok that's a hidden one: after sorting by property if you want to export to phy and also keep the property grouping you have to set the sparsity correctly with the argument `sparsity_dict=dict(method="by_property", by_property="group")`

You can check the `export_to_phy` docs for more info ;)

alejoe91, 2022-03-25:
It seems that the recording doesn't have the group property from the message...can you check?

mfvd, 2022-03-28:
When I check with `rec_cashed.get_property_keys()`, the recording has a group property and the channels have assigned groups with `rec_cashed.get_channel_groups()`.

alejoe91, 2022-04-01:
@mfvd what is the output of:

```
rec_by = rec_cashed.split_by("group")
print(rec_by.keys())
```

?

mfvd, 2022-04-01:
Hi @alejoe91 

The output is:  `dict_keys([0, 1, 2])`.

alejoe91, 2022-04-01:
what about `np.unique(waveforms_ms4.sorting.get_property("group"))`?

mfvd, 2022-04-01:
I get the following: `['0' '1' '2']`

alejoe91, 2022-04-01:
Ah I see! There seems to be a mismatch in the dtype: `groups` for recordings are `int` anf for sorting are `str`..I'll look into it!

alejoe91, 2022-04-02:
@mfvd should be fixed with this PR: https://github.com/SpikeInterface/spikeinterface/pull/579

Can you try?

mfvd, 2022-04-04:
Thanks @alejoe91 
Sorry, I'm quite new to github... how can I install this PR to test it?

closed 2022-04-04 by samuelgarcia

mfvd, 2022-04-04:
I managed to do it and it solved the issue! Thanks!