SpikeInterface/spikeinterface #577: Installed_sorters()

benkiisma, 2022-03-30:
Hello, 

I am running the SpikeInterface_tutorial of August 2021 with the provided dataset (cambridge.bin). I followed the first instructions and set up the environment with the command '**_conda env create -f environment.yml_**'. The first part of the code works perfectly until I ran the 'ss.installed_sorters' command. There, a ValueError appears : '**_numpy.ndarray has the wrong size, try recompiling. Expected 88, got 96_**'. After recompiling, this function output an empty array : [] even if the sorters are installed in the environment. 

Thank you for your help !

samuelgarcia, 2022-03-30:
Hi,
on which system did you installed spikeinterface and sorters ?
sorter are becoming more and more complicated to install in the same environement.
To fix this we are maintenaing some docker/singularity images with sorter installed.
So if you have an ubuntu I would recommand unsintall sorter from the env and play with docker image.

benkiisma, 2022-03-30:
Hi, 
Thank you for your quick answer. I am running on Windows ! For now, I have installed 2 sorters : Klusta and Herdingspikes

benkiisma, 2022-03-30:
I tried to run it on macOS Monterey and still have the same ValueError !!

samuelgarcia, 2022-03-30:
for klusta the only possibility is to have python3.7 which is about to be unmaintain.
have in mind that klusta is unmaintained for long time. And soon it will be impossible to install!
So for klusta the docker is a good option but need linux or macos.

benkiisma, 2022-03-30:
Ok ! And what about herdingspikes ? The same error goes for that sorting algorithm too ?

samuelgarcia, 2022-03-30:
I think herdingspikes made a release recently https://pypi.org/project/herdingspikes/#history
The latest version was not working properly with spikeinetrface.
Maybe try the previsous one.

benkiisma, 2022-03-30:
Thank you, you are right ! Herdingspikes is now working perfectly with the version (0.3.7). However, for the next sorter in the tutorial (IronClust), the following error appear when running : '**_Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json'_**.
The spikeinterface_log.json file contain the following: 
```
{
    "sorter_name": "ironclust",
    "sorter_version": "5.9.8",
    "datetime": "2022-03-30T11:07:49.359292",
    "runtime_trace": [
        "results_IC/run_ironclust.sh: line 3: matlab: command not found"
    ],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 198, in run_from_folder\n    SorterClass._run_from_folder(output_folder, sorter_params, verbose)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/ironclust/ironclust.py\", line 236, in _run_from_folder\n    raise Exception('ironclust returned a non-zero exit code')\nException: ironclust returned a non-zero exit code\n",
    "run_time": null
}
```

samuelgarcia, 2022-03-30:
For matlab based sorter.
The best for debugging is to run in matlab the generated file .m and check the problem there.

benkiisma, 2022-03-30:
As soon as I run the Matlab file ('run_ironclust.m') the Matlab window closes. 
I have another issue with herdingspikes, as soon as I use my data, which is an Intan file I read with the '**_se.read_intan()'_** function, the error '**_Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json_**' appears. I am still running the 0.3.7 version of HerdingSpikes. 
The content of the spikeinterface_log.json file is : 
```
{
    "sorter_name": "herdingspikes",
    "sorter_version": "0.3.7",
    "datetime": "2022-03-30T11:41:33.640831",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 198, in run_from_folder\n    SorterClass._run_from_folder(output_folder, sorter_params, verbose)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/herdingspikes/herdingspikes.py\", line 160, in _run_from_folder\n    recording = st.normalize_by_quantile(\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/normalize_scale.py\", line 175, in normalize_by_quantile\n    return NormalizeByQuantileRecording(*args, **kwargs)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/normalize_scale.py\", line 55, in __init__\n    random_data = get_random_data_chunks(recording, **random_chunk_kwargs)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/utils.py\", line 39, in get_random_data_chunks\n    chunk = recording.get_traces(start_frame=start_frame,\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/baserecording.py\", line 100, in get_traces\n    traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/frameslicerecording.py\", line 66, in get_traces\n    traces = self._parent_recording_segment.get_traces(parent_start, parent_end, channel_indices)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/common_reference.py\", line 117, in get_traces\n    all_traces = self.parent_recording_segment.get_traces(start_frame, end_frame, slice(None))\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/filter.py\", line 68, in get_traces\n    traces_chunk, left_margin, right_margin = get_chunk_with_margin(self.parent_recording_segment,\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/tools.py\", line 31, in get_chunk_with_margin\n    traces_chunk = rec_segment.get_traces(start_frame - left_margin, end_frame + right_margin, channel_indices)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/channelslicerecording.py\", line 77, in get_traces\n    traces = self._parent_recording_segment.get_traces(start_frame, end_frame, parent_indices)\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py\", line 104, in get_traces\n    raw_traces = self.neo_reader.get_analogsignal_chunk(\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/neo/rawio/baserawio.py\", line 580, in get_analogsignal_chunk\n    raw_chunk = self._get_analogsignal_chunk(\n  File \"/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/neo/rawio/intanrawio.py\", line 150, in _get_analogsignal_chunk\n    sigs_chunk = np.zeros((i_stop - i_start, len(channel_names)), dtype='uint16')\nValueError: negative dimensions are not allowed\n",
    "run_time": null
}
```
Thank you for your help.

benkiisma, 2022-03-30:
The same goes for mountainsort4 and tridesclous sorters when using my Intan file. I guess this is because the file I used is not binary. For the tridesclous sorter, I get the following error : '**_'float' object cannot be interpreted as an integer_**' whereas for moutainsort4 I get the exact same error as for herdingspikes. 
Is there a way I can transform my data into binary ?

alejoe91, 2022-03-30:
Could you always share the code that triggered the error and the entire Traceback?

benkiisma, 2022-03-30:
**For IronClust with the cambridge.bin datafile:**
```
sorting_IC = ss.run_ironclust(recording_saved, 
                              output_folder='results_IC',
                              verbose=True)
print(f'IronClust found {len(sorting_IC.get_unit_ids())} units')
```
```
Error running ironclust
Exception: ironclust returned a non-zero exit code

---------------------------------------------------------------------------

SpikeSortingError                         Traceback (most recent call last)
Input In [59], in <cell line: 2>()
      1 # run spike sorting by group
----> 2 sorting_IC = ss.run_ironclust(recording_saved, 
      3                               output_folder='results_IC',
      4                               verbose=True)
      5 print(f'IronClust found {len(sorting_IC.get_unit_ids())} units')

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:301, in run_ironclust(*args, **kwargs)
    300 def run_ironclust(*args, **kwargs):
--> 301     return run_sorter('ironclust', *args, **kwargs)

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:56, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, **sorter_params)
     49 """
     50 Generic function to run a sorter via function approach.
     51 
     52 >>> sorting = run_sorter('tridesclous', recording)
     53 """ + _common_param_doc
     55 if docker_image is None:
---> 56     sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     57                                remove_existing_folder=remove_existing_folder,
     58                                delete_output_folder=delete_output_folder,
     59                                verbose=verbose, raise_error=raise_error, **sorter_params)
     60 else:
     61     sorting = run_sorter_docker(sorter_name, recording, docker_image, output_folder=output_folder,
     62                                 remove_existing_folder=remove_existing_folder,
     63                                 delete_output_folder=delete_output_folder,
     64                                 verbose=verbose, raise_error=raise_error, **sorter_params)

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:80, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, **sorter_params)
     78 SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
     79 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
---> 80 SorterClass.run_from_folder(output_folder, raise_error, verbose)
     81 sorting = SorterClass.get_result_from_folder(output_folder)
     83 if delete_output_folder:

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py:234, in BaseSorter.run_from_folder(cls, output_folder, raise_error, verbose)
    232 if has_error and raise_error:
    233     print(log['error_trace'])
--> 234     raise SpikeSortingError(
    235         "Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
    237 return run_time

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json

```

benkiisma, 2022-03-30:
**For HerdingSpikes with Intan rhd file:** 

```
sorting_HS = ss.run_herdingspikes(recording_sub, output_folder='results_HS', 
                                  filter=False, verbose=True)
print('Found', len(sorting_HS.get_unit_ids()), 'units')
```

```
Error running herdingspikes
Traceback (most recent call last):
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 198, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/herdingspikes/herdingspikes.py", line 160, in _run_from_folder
    recording = st.normalize_by_quantile(
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/normalize_scale.py", line 175, in normalize_by_quantile
    return NormalizeByQuantileRecording(*args, **kwargs)
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/normalize_scale.py", line 55, in __init__
    random_data = get_random_data_chunks(recording, **random_chunk_kwargs)
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/utils.py", line 39, in get_random_data_chunks
    chunk = recording.get_traces(start_frame=start_frame,
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/baserecording.py", line 100, in get_traces
    traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/frameslicerecording.py", line 66, in get_traces
    traces = self._parent_recording_segment.get_traces(parent_start, parent_end, channel_indices)
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/common_reference.py", line 117, in get_traces
    all_traces = self.parent_recording_segment.get_traces(start_frame, end_frame, slice(None))
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/filter.py", line 68, in get_traces
    traces_chunk, left_margin, right_margin = get_chunk_with_margin(self.parent_recording_segment,
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/toolkit/preprocessing/tools.py", line 31, in get_chunk_with_margin
    traces_chunk = rec_segment.get_traces(start_frame - left_margin, end_frame + right_margin, channel_indices)
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/channelslicerecording.py", line 77, in get_traces
    traces = self._parent_recording_segment.get_traces(start_frame, end_frame, parent_indices)
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 104, in get_traces
    raw_traces = self.neo_reader.get_analogsignal_chunk(
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/neo/rawio/baserawio.py", line 580, in get_analogsignal_chunk
    raw_chunk = self._get_analogsignal_chunk(
  File "/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/neo/rawio/intanrawio.py", line 150, in _get_analogsignal_chunk
    sigs_chunk = np.zeros((i_stop - i_start, len(channel_names)), dtype='uint16')
ValueError: negative dimensions are not allowed

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Input In [49], in <cell line: 2>()
      1 # run spike sorting on entire recording
----> 2 sorting_HS = ss.run_herdingspikes(recording_sub, output_folder='results_HS', 
      3                                   filter=False, verbose=True)
      4 print('Found', len(sorting_HS.get_unit_ids()), 'units')

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:343, in run_herdingspikes(*args, **kwargs)
    342 def run_herdingspikes(*args, **kwargs):
--> 343     return run_sorter('herdingspikes', *args, **kwargs)

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:56, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, **sorter_params)
     49 """
     50 Generic function to run a sorter via function approach.
     51 
     52 >>> sorting = run_sorter('tridesclous', recording)
     53 """ + _common_param_doc
     55 if docker_image is None:
---> 56     sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     57                                remove_existing_folder=remove_existing_folder,
     58                                delete_output_folder=delete_output_folder,
     59                                verbose=verbose, raise_error=raise_error, **sorter_params)
     60 else:
     61     sorting = run_sorter_docker(sorter_name, recording, docker_image, output_folder=output_folder,
     62                                 remove_existing_folder=remove_existing_folder,
     63                                 delete_output_folder=delete_output_folder,
     64                                 verbose=verbose, raise_error=raise_error, **sorter_params)

[... 161 lines cut ...]
[... cut ...]
File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/numpy/core/memmap.py:269, in memmap.__new__(subtype, filename, dtype, mode, offset, shape, order)
    266 array_offset = offset - start
    267 mm = mmap.mmap(fid.fileno(), bytes, access=acc, offset=start)
--> 269 self = ndarray.__new__(subtype, shape, dtype=descr, buffer=mm,
    270                        offset=array_offset, order=order)
    271 self._mmap = mm
    272 self.offset = offset

TypeError: 'float' object cannot be interpreted as an integer
```

alejoe91, 2022-04-01:
@benkiisma can you share the entire code? 

You can also try to save the recording before spike sorting:

`rec = rec_processed.save(folder="preprocessed")`

benkiisma, 2022-04-01:
Thank you for your answer. When running the line you are suggesting, I have an error : 
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [46], in <cell line: 1>()
----> 1 rec = recording_sub.save(folder="preprocessed")

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/base.py:519, in BaseExtractor.save(self, **kwargs)
    517     return self.save_to_memory(**kwargs)
    518 else:
--> 519     return self.save_to_folder(**kwargs)

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/base.py:594, in BaseExtractor.save_to_folder(self, name, folder, dump_ext, verbose, **save_kwargs)
    587     provenance_file.write_text(
    588         json.dumps({'warning': 'the provenace is not dumpable!!!'}),
    589         encoding='utf8'
    590     )
    593 # save data (done the subclass)
--> 594 cached = self._save(folder=folder, verbose=verbose, **save_kwargs)
    596 self.save_metadata_to_folder(folder)
    598 # copy properties/

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/baserecording.py:142, in BaseRecording._save(self, format, **save_kwargs)
    139 dtype = save_kwargs.get('dtype', 'float32')
    141 job_kwargs = {k: save_kwargs[k] for k in self._job_keys if k in save_kwargs}
--> 142 write_binary_recording(self, file_paths=file_paths, dtype=dtype, **job_kwargs)
    144 from .binaryrecordingextractor import BinaryRecordingExtractor
    145 cached = BinaryRecordingExtractor(file_paths=file_paths, sampling_frequency=self.get_sampling_frequency(),
    146                                   num_chan=self.get_num_channels(), dtype=dtype,
    147                                   channel_ids=self.get_channel_ids(), time_axis=0,
    148                                   file_offset=0, gain_to_uV=self.get_channel_gains(),
    149                                   offset_to_uV=self.get_channel_offsets())

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/core_tools.py:226, in write_binary_recording(recording, file_paths, dtype, add_file_extension, verbose, byte_offset, **job_kwargs)
    224 file_path = file_paths[segment_index]
    225 shape = (num_frames, num_channels)
--> 226 rec_memmap = np.memmap(str(file_path), dtype=dtype, mode='w+', offset=byte_offset, shape=shape)
    227 rec_memmaps.append(rec_memmap)
    228 rec_memmaps_dict.append(dict(filename=str(file_path), dtype=dtype, mode='r+', offset=byte_offset, shape=shape))

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/numpy/core/memmap.py:269, in memmap.__new__(subtype, filename, dtype, mode, offset, shape, order)
    266 array_offset = offset - start
    267 mm = mmap.mmap(fid.fileno(), bytes, access=acc, offset=start)
--> 269 self = ndarray.__new__(subtype, shape, dtype=descr, buffer=mm,
    270                        offset=array_offset, order=order)
    271 self._mmap = mm
    272 self.offset = offset

TypeError: 'float' object cannot be interpreted as an integer
```
**Here is my entire code:** 

```
import spikeinterface as si
import spikeinterface.extractors as se 
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw

import matplotlib.pyplot as plt
import seaborn as sns
import numpy as np

%matplotlib inline

# file path
recording_file = '3782_REF_week2_06102022_211006_152133.rhd'

# parameters to load the bin/dat format
[... 75 lines cut ...]
                            5,19,1,29,14,28,18,12,2,30,17,31,16,0,15])
probe.set_device_channel_indices(channel_indices)

recording_prb = recording.set_probe(probe)

recordings_by_group = recording_prb.split_by("group")
recording_to_process = recordings_by_group[0]

recording_f = st.bandpass_filter(recording_to_process, freq_min=300, freq_max=6000)

recording_cmr = st.common_reference(recording_f, reference='global', operator='median')

fs = recording_cmr.get_sampling_frequency()
recording_sub = recording_cmr.frame_slice(start_frame=0*fs, end_frame=300*fs)

sorting_HS = ss.run_herdingspikes(recording_sub, output_folder='results_HS', 
                                  filter=False, verbose=True)
print('Found', len(sorting_HS.get_unit_ids()), 'units')

sorting_MS = ss.run_mountainsort4(recording_sub)
print('Found', len(sorting_MS.get_unit_ids()), 'units')

sorting_TS = ss.run_tridesclous(recording_sub)
print('Found', len(sorting_TS.get_unit_ids()), 'units')
```

benkiisma, 2022-04-02:
As you proposed, when I saved, I managed to run 2 sorters: Mountainsort4 and tridesclous. Herdingspikes still don't work and I can't figure out why. However, Mountainsort4 and tridesclous output warnings even if the code runs .

**I have a first warning when I run : ss.installed_sorters()**

`/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/ironclust/ironclust.py:152: ResourceWarning: unclosed file <_io.TextIOWrapper name=74 encoding='UTF-8'>
  if check_compiled():
ResourceWarning: Enable tracemalloc to get the object allocation traceback`

**Then, I have an error when running herdingspikes**

**Code for Herdingspikes**

```
sorting_HS = ss.run_herdingspikes(recording_loaded, output_folder='results_HS', 
                                  filter=False, verbose=True)
print('Found', len(sorting_HS.get_unit_ids()), 'units')
```

**Error for Herding Spike**

`herdingspikes version<0.3.99 uses the OLD spikeextractors with NewToOldRecording.
Consider updating herdingspikes (pip install herdingspikes>=0.3.99)
# Generating new position and neighbor files from data file
# Not Masking any Channels
# Sampling rate: 30000
# Localization On
# Number of recorded channels: 32
# Analysing frames: 4428544; Seconds: 147.61813333333333
# Frames before spike in cutout: 9
# Frames after spike in cutout: 54
# tcuts: 39 84
# tInc: 100000
# Analysing frames from -39 to 100084  (0.0%)
# Analysing frames from 99961 to 200084  (2.3%)
# Analysing frames from 199961 to 300084  (4.5%)
# Analysing frames from 299961 to 400084  (6.8%)
# Analysing frames from 399961 to 500084  (9.0%)
# Analysing frames from 499961 to 600084  (11.3%)
# Analysing frames from 599961 to 700084  (13.5%)
# Analysing frames from 699961 to 800084  (15.8%)
# Analysing frames from 799961 to 900084  (18.1%)
# Analysing frames from 899961 to 1000084  (20.3%)
# Analysing frames from 999961 to 1100084  (22.6%)
# Analysing frames from 1099961 to 1200084  (24.8%)
# Analysing frames from 1199961 to 1300084  (27.1%)
# Analysing frames from 1299961 to 1400084  (29.4%)
# Analysing frames from 1399961 to 1500084  (31.6%)
# Analysing frames from 1499961 to 1600084  (33.9%)
# Analysing frames from 1599961 to 1700084  (36.1%)
# Analysing frames from 1699961 to 1800084  (38.4%)
# Analysing frames from 1799961 to 1900084  (40.6%)
# Analysing frames from 1899961 to 2000084  (42.9%)
# Analysing frames from 1999961 to 2100084  (45.2%)
# Analysing frames from 2099961 to 2200084  (47.4%)
# Analysing frames from 2199961 to 2300084  (49.7%)
# Analysing frames from 2299961 to 2400084  (51.9%)
# Analysing frames from 2399961 to 2500084  (54.2%)
# Analysing frames from 2499961 to 2600084  (56.5%)
# Analysing frames from 2599961 to 2700084  (58.7%)
# Analysing frames from 2699961 to 2800084  (61.0%)
# Analysing frames from 2799961 to 2900084  (63.2%)
# Analysing frames from 2899961 to 3000084  (65.5%)
# Analysing frames from 2999961 to 3100084  (67.7%)
# Analysing frames from 3099961 to 3200084  (70.0%)
# Analysing frames from 3199961 to 3300084  (72.3%)
# Analysing frames from 3299961 to 3400084  (74.5%)
# Analysing frames from 3399961 to 3500084  (76.8%)
# Analysing frames from 3499961 to 3600084  (79.0%)
# Analysing frames from 3599961 to 3700084  (81.3%)
# Analysing frames from 3699961 to 3800084  (83.5%)
[... 65 lines cut ...]
**Then I have a warning when I run mountainsort4:**

`sorting_MS = ss.run_mountainsort4(recording_loaded)
print('Found', len(sorting_MS.get_unit_ids()), 'units')`

`Mountainsort4 use the OLD spikeextractors mapped with NewToOldRecording
/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/klusta/__init__.py:42: ResourceWarning: unclosed file <_io.TextIOWrapper name='/dev/null' mode='w' encoding='UTF-8'>
  __version_git__ = __version__ + _git_version()
ResourceWarning: Enable tracemalloc to get the object allocation traceback
/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikemetrics/metrics.py:15: DeprecationWarning: Please use `gaussian_filter1d` from the `scipy.ndimage` namespace, the `scipy.ndimage.filters` namespace is deprecated.
  from scipy.ndimage.filters import gaussian_filter1d
/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/klusta/__init__.py:42: ResourceWarning: unclosed file <_io.TextIOWrapper name='/dev/null' mode='w' encoding='UTF-8'>
  __version_git__ = __version__ + _git_version()
ResourceWarning: Enable tracemalloc to get the object allocation traceback
/Users/ismail/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikemetrics/metrics.py:15: DeprecationWarning: Please use `gaussian_filter1d` from the `scipy.ndimage` namespace, the `scipy.ndimage.filters` namespace is deprecated.
  from scipy.ndimage.filters import gaussian_filter1d`

**I have again a warning when I run tridesclous **

`OMP: Info #271: omp_set_nested routine deprecated, please use omp_set_max_active_levels instead.`

I guess that the warnings are not from my code but from the sorter's code !! 
How can I make herdingspikes work ?

Thank you very much !!

alejoe91, 2022-04-02:
@benkiisma don't worry about the warnings. For HS we recommend asking on the herding spikes GitHub repo (https://github.com/mhhennig/HS2) as the sorters are separate packages!

benkiisma, 2022-04-02:
Thank you very much @alejoe91 for your help !!

alejoe91, 2022-04-02:
You are welcome! Closing the issue!

closed 2022-04-02 by alejoe91