SpikeInterface/spikeinterface #580: Trouble running Spyking-Circus through Spikeinterface but not through Anaconda Terminal

Erik2020-Neuronz, 2022-04-01:
Hello All,

I was able to successfully spike sort my dataset using Kilosort2 and Tridesclous, but ran into some trouble with Spyking-Circus. It began sorting with Circus, but kicked out soon into the process. I attached the traceback from pycharm in "CricusTraceback_Si_env.txt" below. 
I've also attached the spykingcircus log and the spikeinterface log below. The probe file it returned is correct and is the same file output by ks2 and Tride (checked ks2 output in phy and it looked good). Spikeinterface and the sorters are being run from a conda environment made from the "full_spikeinterface_environment_windows.yml" from the installation_tips section on Github (attached below). If I run spyking-cricus directly from an anaconda terminal while still in the same anaconda environment (si_env) then it completes the sorting fine. I've attached the code below

```
import spikeinterface.extractors as se
import spikeinterface.sorters as ss
import probeinterface as pi
data_file = 'D:\\GroundTruth\\m15_190315_145422_EC\\m15_190315_145422_EC.dat'
recording = se.BinaryRecordingExtractor(data_file, 30000, 32, 'int16', None, None, 0, 0, None, None, None)
probegroup = pi.read_prb("Poly3_Chronic.prb") 
recording_prb = recording.set_probegroup(probegroup)

ss.Kilosort2Sorter.set_kilosort2_path(r'C:\Users\Erik\OneDrive - Virginia Tech\Documents\GitHub\Kilosort-2.0\Kilosort-2.0')
kilosort_output = 'D:\\GroundTruth\\m15_190315_145422_EC\\kilosort_si_env'
sorting_Kilosort = ss.run_kilosort2(recording_prb, output_folder=kilosort_output, verbose=True)

tride_output = 'D:\\GroundTruth\\m15_190315_145422_EC\\Tride_si_env'
sorting_Tride = ss.run_tridesclous(recording_prb, output_folder=tride_output, verbose=True)

circus_output = 'D:\\GroundTruth\\m15_190315_145422_EC\\Circus_si_env'
sorting_Circus = ss.run_spykingcircus(recording_prb, output_folder=circus_output, verbose=True)
```

where Poly3_Chronic.prb is the probe file it reads from disk. Any help would be appreciated, thanks in advance!

also:
I am on a Windows 10 machine, Intel I-7 cpu and Nvidia Graphics card. Using Pycharm to run the python scripts for spikeinterface, and using Anaconda3 as the package manager for the conda environment 

[CricusTraceback_Si_env.txt](https://github.com/SpikeInterface/spikeinterface/files/8401090/CricusTraceback_Si_env.txt)
[spykingcircus.log.txt](https://github.com/SpikeInterface/spikeinterface/files/8401091/spykingcircus.log.txt)
[full_spikeinterface_environment_windows.yml.txt](https://github.com/SpikeInterface/spikeinterface/files/8401092/full_spikeinterface_environment_windows.yml.txt)
[spikeinterface_log.json.txt](https://github.com/SpikeInterface/spikeinterface/files/8401093/spikeinterface_log.json.txt)

alejoe91, 2022-04-20:
Hi @Erik2020-Neuronz 

Can you ask for help on the Spyking circus github repo? Probably the problem is related to some installation issues

Erik2020-Neuronz, 2022-04-21:
Sounds good,
Thank you for the response

closed 2022-04-21 by Erik2020-Neuronz