SpikeInterface/spikeinterface #581: Running `sc.compare_two_sorters` when one sorting contains units with a single spike throws error

khl02007, 2022-04-02:
Seems like `sorting.get_unit_spike_train` returns just an int rather than a list when the unit has only one spike. The sortings are done with kilosort3 and loaded with `se.read_kilosort`.

@rly @alejoe91 

Error:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-5-5cce6323a10a> in <module>
      2                               sorting2=sorting_di,
      3                               sorting1_name='Raw',
----> 4                               sorting2_name='DI')

~/.local/lib/python3.7/site-packages/spikeinterface/comparison/paircomparisons.py in compare_two_sorters(*args, **kwargs)
    148 
    149 def compare_two_sorters(*args, **kwargs):
--> 150     return SymmetricSortingComparison(*args, **kwargs)
    151 
    152 

~/.local/lib/python3.7/site-packages/spikeinterface/comparison/paircomparisons.py in __init__(self, sorting1, sorting2, sorting1_name, sorting2_name, delta_time, sampling_frequency, match_score, chance_score, n_jobs, verbose)
    116                                           delta_time=delta_time,
    117                                           match_score=match_score, chance_score=chance_score,
--> 118                                           n_jobs=n_jobs, verbose=verbose)
    119 
    120     def get_matching(self):

~/.local/lib/python3.7/site-packages/spikeinterface/comparison/paircomparisons.py in __init__(self, sorting1, sorting2, sorting1_name, sorting2_name, delta_time, match_score, chance_score, n_jobs, verbose)
     34 
     35 
---> 36         self._do_agreement()
     37         self._do_matching()
     38 

~/.local/lib/python3.7/site-packages/spikeinterface/comparison/paircomparisons.py in _do_agreement(self)
     59         # common to GroundTruthComparison and SymmetricSortingComparison
     60         # spike count for each spike train
---> 61         self.event_counts1 = do_count_event(self.sorting1)
     62         self.event_counts2 = do_count_event(self.sorting2)
     63 

~/.local/lib/python3.7/site-packages/spikeinterface/comparison/comparisontools.py in do_count_event(sorting)
     78     """
     79     unit_ids = sorting.get_unit_ids()
---> 80     ev_counts = np.array([len(sorting.get_unit_spike_train(u)) for u in unit_ids], dtype='int64')
     81     event_counts = pd.Series(ev_counts, index=unit_ids)
     82     return event_counts

~/.local/lib/python3.7/site-packages/spikeinterface/comparison/comparisontools.py in <listcomp>(.0)
     78     """
     79     unit_ids = sorting.get_unit_ids()
---> 80     ev_counts = np.array([len(sorting.get_unit_spike_train(u)) for u in unit_ids], dtype='int64')
     81     event_counts = pd.Series(ev_counts, index=unit_ids)
     82     return event_counts

TypeError: len() of unsized object
```

alejoe91, 2022-04-05:
Thanks for the report @khl02007 

Do you have time to look for a fix?

khl02007, 2022-04-05:
@alejoe91 sure I will try to submit a PR soon

alejoe91, 2022-04-20:
Probably related to https://github.com/SpikeInterface/spikeinterface/issues/601

alejoe91, 2022-05-03:
@khl02007 this is using kilosort right?

khl02007, 2022-05-04:
@alejoe91 yes it is. sorry for being slow on this issue. how do you think this should be fixed? seems like we should make kilosort extractor return only the units that have nonzero number of spikes?

alejoe91, 2022-05-05:
> @alejoe91 yes it is. sorry for being slow on this issue. how do you think this should be fixed? seems like we should make kilosort extractor return only the units that have nonzero number of spikes?

I think we should make all sorters do that! We have a new function `sorting.remove_empty_units()` that selects only the units with nspikes>0. For spiketrains with 1 spike, the issue should be solved with https://github.com/SpikeInterface/spikeinterface/pull/617

closed 2022-08-01 by alejoe91