SpikeInterface/spikeinterface #583: Object of type SubRecordingExtractor is not JSON serializable

lkeegan, 2022-04-04:
If I concatenate multiple recordings converted using `create_recording_from_old_extractor`:
```
recordings = []
for file in smrx_files:
    # Automatically select Rhd channels
    channel_info = legacy_se.CEDRecordingExtractor.get_all_channels_info(file)
    rhd_channels = []
    rhd_channel_idxs = []
    for idx, (ch, info) in enumerate(channel_info.items()):
        if "Rhd" in info["title"]:
            rhd_channels.append(ch)
            rhd_channel_idxs.append(idx)
    rec = legacy_se.CEDRecordingExtractor(file, smrx_channel_ids=rhd_channels)
    rec = legacy_se.load_probe_file(rec, probe_file)
    
    # convert to new API recording
    rec = si.create_recording_from_old_extractor(rec)
    recordings.append(rec)

# new API: concatenate all recordings to form a single long recording                                                                                                                                       
multi_recording = si.concatenate_recordings(recordings)
```
When I try to run spike sorting on this recording I get this error:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [124], in <module>
----> 1 sorting_outputs = ss.run_sorters(
      2     sorter_list=sorter_list, 
      3     working_folder=spikeinterface_folder / 'simple_bin_output',
      4     recording_dict_or_list=dict(rec0=recording_processed), 
      5     sorter_params=sorter_params,
      6     mode_if_folder_exists="overwrite", # "overwrite" to overwrite # change to "keep" to avoid repeating the spike sorting
      7     verbose=True,
      8 )

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/sorters/launcher.py:257, in run_sorters(sorter_list, recording_dict_or_list, working_folder, sorter_params, mode_if_folder_exists, engine, engine_kwargs, verbose, with_output, docker_images, singularity_images)
    254 if engine == 'loop':
    255     # simple loop in main process
    256     for task_args in task_args_list:
--> 257         _run_one(task_args)
    259 elif engine == 'joblib':
    260     from joblib import Parallel, delayed

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/sorters/launcher.py:33, in _run_one(arg_list)
     30 # because we won't want the loop/worker to break
     31 raise_error = False
---> 33 run_sorter(sorter_name, recording, output_folder=output_folder,
     34            remove_existing_folder=remove_existing_folder,
     35            delete_output_folder=delete_output_folder,
     36            verbose=verbose, raise_error=raise_error,
     37            docker_image=docker_image, singularity_image=singularity_image,
     38            with_output=with_output, **sorter_params)

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:67, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
     61 if singularity_image is not None:
     62     return run_sorter_container(sorter_name, recording, 'singularity', singularity_image,
     63                                 output_folder=output_folder,
     64                                 remove_existing_folder=remove_existing_folder,
     65                                 delete_output_folder=delete_output_folder, verbose=verbose,
     66                                 raise_error=raise_error, with_output=with_output, **sorter_params)
---> 67 return run_sorter_local(sorter_name, recording, output_folder=output_folder,
     68                         remove_existing_folder=remove_existing_folder,
     69                         delete_output_folder=delete_output_folder,
     70                         verbose=verbose, raise_error=raise_error, with_output=with_output,
     71                         **sorter_params)

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:84, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     81 SorterClass = sorter_dict[sorter_name]
     83 # only classmethod call not instance (stateless at instance level but state is in folder)
[... 86 lines cut ...]
    404             chunks = _iterencode(value, _current_indent_level)
--> 405         yield from chunks
    406 if newline_indent is not None:
    407     _current_indent_level -= 1

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/json/encoder.py:438, in _make_iterencode.<locals>._iterencode(o, _current_indent_level)
    436         raise ValueError("Circular reference detected")
    437     markers[markerid] = o
--> 438 o = _default(o)
    439 yield from _iterencode(o, _current_indent_level)
    440 if markers is not None:

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/json/encoder.py:179, in JSONEncoder.default(self, o)
    160 def default(self, o):
    161     """Implement this method in a subclass such that it returns
    162     a serializable object for ``o``, or calls the base implementation
    163     (to raise a ``TypeError``).
   (...)
    177 
    178     """
--> 179     raise TypeError(f'Object of type {o.__class__.__name__} '
    180                     f'is not JSON serializable')

TypeError: Object of type SubRecordingExtractor is not JSON serializable
```

alejoe91, 2022-04-05:
@lkeegan I believe this is due to the fact that your old recording is a `SubRecordingExtractor`. I recommend loading the old recording, converting it to new, and then slice it with `frame_slice` or `channel_slice` to avoid this behavior.

lkeegan, 2022-04-05:
@alejoe91 thanks - I'm not sure how to avoid getting a `SubRecordingExtractor` from `CEDRecordingExtractor`, but with `read_ced` and `channel_slice` I can do everything with the new API so I don't actually need to convert the old recording.

closed 2022-04-05 by lkeegan

alejoe91, 2022-04-05:
Yeah if you can go full on new API that's highly recommended! :)