SpikeInterface/spikeinterface #586: performance hit with NWB + fancy indexing

miketrumpis, 2022-04-08:
These lines can lead to slow reads from the NWB extractor due to [fancy indexing](https://docs.h5py.org/en/latest/high/dataset.html#fancy-indexing). Whenever I do something like this, I read the full channel column and then do a fast fancy index from the ndarray in memory. It is technically wasteful, but the use case for memory mapping is to pull out relatively small chunks.

https://github.com/SpikeInterface/spikeinterface/blob/29a64a2f5024cdd11c127159c58ab84facd23af5/spikeinterface/extractors/nwbextractors.py#L226-L238

I refactored this downstream as the following. **Big** improvement when working with a `ChannelSliceRecording`

```python
        if isinstance(channel_indices, slice):
            traces = es.data[start_frame:end_frame, channel_indices]
        else:
            # channel_indices is np.ndarray
            channel_indices = np.asarray(channel_indices)
            if channel_indices.size > 1:
                traces = es.data[start_frame:end_frame][:, channel_indices]
            else:
                traces = es.data[start_frame:end_frame, channel_indices[0]]
```

alejoe91, 2022-04-08:
Hi @miketrumpis 

Have you tested this with unsorted channel_ids? Eg `rec.get_traces(channel_ids=[4, 2, 6])`

samuelgarcia, 2022-04-08:
Hi Mike,
I agree with Alessio when channel_indices is list or index and not ordered it will fail.

The only diff in your case is when there is one channel no ?
```traces = es.data[start_frame:end_frame, channel_indices[0]]```
In our API we must return 2d even for one channel  so we need to reshape (nsample, 1).

miketrumpis, 2022-04-08:
No, not tested. But the first slice `es.data[start_frame:end_frame]` yields an ndarray and the second slice `[:, channel_indices]` has been tested implicitly by numpy.

Thanks for pointing out the second point, I didn't know!

miketrumpis, 2022-04-09:
My bad.. didn't know about that layer of abstraction. Can I rely on [this](https://github.com/catalystneuro/lazy_ops/blob/master/lazy_ops/lazy_loading.py)?

```python
es.data[start_frame:end_frame].dsetread()[:, channel_indices]
```

alejoe91, 2022-04-09:
This approach would load all channels in memory and then slice the one you needs. Let's say you want 60 s from 2 neuropixels channels (saved to NWB). You would need to allocate in RAM 192 times more (for a brief period) than the current approach..depending on the use case, I think you might risk to run out of RAM..but we can discuss about it!

miketrumpis, 2022-04-10:
Excellent points. Certainly 100x memory waste is unacceptable, and probably slower. I will def keep using a subclass for the scenario with <<2x memory overhead and 4x-8x speedup. The not-very-helpful proposal that occurs to me here is a threshold policy (e.g. over/under 2x overhead) for automatically deciding between the two approaches.

alejoe91, 2022-04-20:
@miketrumpis that could be an option indeed!

closed 2022-08-01 by alejoe91