SpikeInterface/spikeinterface #587: Reading raw TDT data

TdSNielsen, 2022-04-08:
I am having problems figuring out how to use the extractors sub-package to import TDT data. In the attached example, I am trying to load data from a specific tank. In the code, folder_path points to the folder representing this tank (i.e., a folder containing all (64) channels and header files). 'eNeu1' is the stream that I wish to load, I have also tried without this argument, which throws the same error. As you can see, the code is not working - I am wondering if it is because I haven't defined "self", but I could not find documentation on this issue...

[image]

samuelgarcia, 2022-04-08:
Hi,
could you also post this to python-neo repo ?
I guess the error is there.
@JuliaSprenger : implemented multi stream recently and should be able to help you.

TdSNielsen, 2022-04-08:
I have posted it there as well, thank you for the input @SamuelGarcia!

TdSNielsen, 2022-05-07:
So, I figured out what caused this problem:

If the TDT data files are located in ~/folder_1/folder_2/, the the filename of the *.Tbk file must be equal to: folder_1_folder_2.Tbk
This is a bit weird, and is not the case when using TDT's own python functions. However, now that I know this, I was able to correct it.

I am still struggling with another issue, though: If I don't specify stream_id, I get an error that multiple streams are available (['0', '1', '2', '3', '4' ]) and I have to choose one of these. This numerical indexing of the stream is not intuitive since it is difficult to know in which order the recording objects are indexed. When I do set stream_id to any of these values, I get another error further downstream (this time originating from baserecording.py (line 31, data type "<class 'int'>" not understood) instead of neobaseextractor.py). 
Besides this, it does not seem to be possible to provide any additional input arguments (like which channel(s) or time window to import, which is necessary in my case since each of my recordings are >100 GB...

Cheers,
Thomas

samuelgarcia, 2022-05-07:
Hi Thomas,
for spikeinterface when the format is multistream on neo side you indeed to specify which one because Recording object is mono stream.
I think for tdt stream are order by descending sampling rate.
Otherwise you can also loop over then inspect channel names and properties until you find the good one.

TdSNielsen, 2022-05-07:
Thanks @samuelgarcia, that makes sense. I have indeed been mainly trying with stream='0' because the relevant stream (store='RSn1' in TDT terminology) is first in the TDT header file. Now I just need to figure out why it throws the data type error on dtype in baserecording.py...

Do you know if it is possible to limit what data gets loaded from a stream? The relevant stream contains 64 channels, of which only 32 are relevant, and the recordings are >5 hours long, so I unfortunately wouldn't be able to load all to memory simultaneously...

TdSNielsen, 2022-07-15:
I never managed to get Neo to work with the TDT data format directly, but I ended up exporting my data to float32 binary format, which seems to work.

closed 2022-07-15 by TdSNielsen

PaolaGith, 2022-07-18:
> Thanks @samuelgarcia, that makes sense. I have indeed been mainly trying with stream='0' because the relevant stream (store='RSn1' in TDT terminology) is first in the TDT header file. Now I just [...]

Hi, may I ask how did you manage to upload the streamer (Rsn) data? I have never used spikeinterface before, but mountainsort directly. So usually I convert the TDT, both the "data tank" and the "streamer" data into matlab data and I create the .mda file as required by mountainsort. If I want to use spikeinterface, once I have my raw TDT data, in the folder I have 64 different files for the neural channels and 7 additional files (tdx, tbk, etc) how should I proceed? 
I also need to cut my signal, since I am only interested in the activity pre and post stimulation. Can I cut the signal based on the stimulation onset and offset?

TdSNielsen, 2022-07-19:
Hi @PaolaGith, 

on this page you can find some information about different ways of converting to binary format (from either Python or Matlab):
https://www.tdt.com/docs/sdk/offline-data-analysis/data-conversion/

I used the Matlab script in the bottom because it enabled exporting the full float-32 information, rather than converting to int-16. Notice that if you use their script, there is an error in the while statement: I changed the initialization to _while T2 < TIME_DELTA*ceil(dur/TIME_DELTA)_, which fixed the error (you will, of course, need to modify the script to feed your own data in which ever way you wish)...

About how to deal with snippet data, I'm not sure (I also have stimulation in my data, but I'm interested in spike sorting of the continuous signal). The binary file that is created does not contain information about stimulation timing, but it must be possible to feed this information somehow - maybe @samuelgarcia knows? (Since you have full control over the way the binary files are saved, you can of course save each sweep as a separate file, but I doubt that would be practical.)

Cheers, 
Thomas

PaolaGith, 2022-07-20:
> Hi @PaolaGith,
>
> on this page you can find some information about different ways of converting to binary format (from either Python or Matlab): [...]

Thanks Thomas! I already converted my data using TDTbin2mat so that I could use it with mountainsort. Does it makes sense to redo the convertion starting from the raw TDT data or can I convert the matlab structure? (sorry to bother you. I am new to python so I get lost sometimes)
According to the Spikeinterface documentation, BaseEvent module should retrieve events from files.

TdSNielsen, 2022-07-20:
Well, if you already have your data in a file format that can be read by the extractors module of SpikeInterface, that is fine. :)

alejoe91, 2022-07-20:
For the stimulation artifacts, if you have the sample indices when the stimulation pulses are triggered, you can use the `preprocrssing.remove_artifacts()` function to blank out an interval at the occurrence of the pulse. I also plan to implement an average template removal soon method that should work well for electrical and opto stimulation.

PaolaGith, 2022-07-21:
> Well, if you already have your data in a file format that can be read by the extractors module of SpikeInterface, that is fine. :)

Thanks again. I thought I could use the spikeinterface upoloading directly the TDT data as they come out from the streamer (the .sev files), avoiding the preprocessing stage with Matlab, but now I understand that I can't skip the preprocessing stage, and I still need to convert the data first. Since, for the data I already collected, I have the .mat files obtained with the Tdtbin2mat function, I will use a function from scipy to load the .mat file and read it with se.NumpyRecording.