SpikeInterface/spikeinterface #589: nearest_neighbors_isolation metric cannot be calculated for sort groups with units with fewer than seven spikes

jguides, 2022-04-09:
I get the following error when I try to calculate the nearest neighbors isolation score for a sort group that has at least one unit with fewer than seven spikes: 

`---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [25], in <cell line: 1>()
     86     metric_key['waveform_params_name'] = waveform_params_name
     87     ss.MetricSelection.insert1(metric_key, skip_duplicates=True)
---> 88     ss.QualityMetrics.populate([(ss.MetricSelection & metric_key).proj()])
     90 # automatic curation
     91 ss.AutomaticCurationParameters().insert_default()

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/datajoint/autopopulate.py:229, in AutoPopulate.populate(self, suppress_errors, return_exception_objects, reserve_jobs, order, limit, max_calls, display_progress, processes, make_kwargs, *restrictions)
    225 if processes == 1:
    226     for key in (
    227         tqdm(keys, desc=self.__class__.__name__) if display_progress else keys
    228     ):
--> 229         error = self._populate1(key, jobs, **populate_kwargs)
    230         if error is not None:
    231             error_list.append(error)

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/datajoint/autopopulate.py:281, in AutoPopulate._populate1(self, key, jobs, suppress_errors, return_exception_objects, make_kwargs)
    279 self.__class__._allow_insert = True
    280 try:
--> 281     make(dict(key), **(make_kwargs or {}))
    282 except (KeyboardInterrupt, SystemExit, Exception) as error:
    283     try:

File ~/Src/spyglass/src/spyglass/spikesorting/spikesorting_curation.py:415, in QualityMetrics.make(self, key)
    413 params = (MetricParameters & key).fetch1('metric_params')
    414 for metric_name, metric_params in params.items():
--> 415     metric = self._compute_metric(
    416         waveform_extractor, metric_name, **metric_params)
    417     qm[metric_name] = metric
    418 qm_name = self._get_quality_metrics_name(key)

File ~/Src/spyglass/src/spyglass/spikesorting/spikesorting_curation.py:454, in QualityMetrics._compute_metric(self, waveform_extractor, metric_name, **metric_params)
    452     metric = {}
    453     for unit_id in waveform_extractor.sorting.get_unit_ids():
--> 454         metric[str(unit_id)] = metric_func(
    455             waveform_extractor, this_unit_id=unit_id, **metric_params)
    456 return metric

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/spikeinterface/toolkit/qualitymetrics/pca_metrics.py:378, in nearest_neighbors_isolation(waveform_extractor, this_unit_id, max_spikes_for_nn, n_neighbors, n_components, radius_um, seed)
    375 all_snippets = np.concatenate([waveforms_target_unit_sampled.reshape((n_snippets, -1)),
    376                                waveforms_other_unit_sampled.reshape((n_snippets, -1))], axis=0)
    377 pca = IncrementalPCA(n_components=n_components)
--> 378 pca.partial_fit(all_snippets)
    379 projected_snippets = pca.transform(all_snippets)
    381 # compute isolation

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/sklearn/decomposition/_incremental_pca.py:283, in IncrementalPCA.partial_fit(self, X, y, check_input)
    277     raise ValueError(
    278         "n_components=%r invalid for n_features=%d, need "
    279         "more rows than columns for IncrementalPCA "
    280         "processing" % (self.n_components, n_features)
    281     )
    282 elif not self.n_components <= n_samples:
--> 283     raise ValueError(
    284         "n_components=%r must be less or equal to "
    285         "the batch number of samples "
    286         "%d." % (self.n_components, n_samples)
    287     )
    288 else:
    289     self.n_components_ = self.n_components

ValueError: n_components=7 must be less or equal to the batch number of samples 2.
`
Thanks in advance for help with addressing this.

alejoe91, 2022-04-09:
Thanks for the report. I think we should check first and discard units with less than n_neighbors spikes

khl02007, 2022-04-09:
@alejoe91 I agree. How about 
1. discard units with less than n_neighbors spikes
2. compute nearest neighbor metrics for the others
3. return the computed metrics, but also return `nan` for the units that were discarded

I can work on this, let me know what you think

alejoe91, 2022-04-10:
Thanks @khl02007 

Additionally I would add a warning when these low count units are found!! If you can make a PR it would be great!

alejoe91, 2022-04-20:
Fixed here: https://github.com/SpikeInterface/spikeinterface/pull/591

closed 2022-04-20 by alejoe91