SpikeInterface/spikeinterface #592: Error when concatenating recordings in new spike interface

jguides, 2022-04-11:
Our lab previously used the recording_files attribute to concatenate recordings, but this seems to be gone in the new version of spike interface. This leads to an error when we try to concatenate recordings, for example as below. 
--> 166     recording = si.concatenate_recordings(recording.recording_list).
Could you recommend a strategy for concatenating recordings in the new spike interface? Thanks in advance for any help. 

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Input In [15], in <cell line: 1>()
     24     artifact_key['artifact_params_name'] = parameter_set_dict["artifact"][sort_group_brain_region]
     25     nd.spikesorting.ArtifactDetectionSelection.insert1(artifact_key, skip_duplicates=True)
---> 26     nd.spikesorting.ArtifactDetection.populate([(nd.spikesorting.ArtifactDetectionSelection & artifact_key).proj()])
     27     artifact_removed_interval_list_name = (nd.spikesorting.ArtifactDetection & artifact_key).fetch1('artifact_removed_interval_list_name')
     28 # Detect artifacts across sort groups for tetrodes

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/datajoint/autopopulate.py:229, in AutoPopulate.populate(self, suppress_errors, return_exception_objects, reserve_jobs, order, limit, max_calls, display_progress, processes, make_kwargs, *restrictions)
    225 if processes == 1:
    226     for key in (
    227         tqdm(keys, desc=self.__class__.__name__) if display_progress else keys
    228     ):
--> 229         error = self._populate1(key, jobs, **populate_kwargs)
    230         if error is not None:
    231             error_list.append(error)

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/datajoint/autopopulate.py:281, in AutoPopulate._populate1(self, key, jobs, suppress_errors, return_exception_objects, make_kwargs)
    279 self.__class__._allow_insert = True
    280 try:
--> 281     make(dict(key), **(make_kwargs or {}))
    282 except (KeyboardInterrupt, SystemExit, Exception) as error:
    283     try:

File ~/Src/spyglass/src/spyglass/spikesorting/spikesorting_artifact.py:74, in ArtifactDetection.make(self, key)
     71 recording_name = SpikeSortingRecording._get_recording_name(key)
     72 recording = si.load_extractor(recording_path)
---> 74 artifact_removed_valid_times, artifact_times = _get_artifact_times(
     75     recording, **artifact_params)
     77 # NOTE: decided not to do this but to just create a single long segment; keep for now
     78 # get artifact times by segment
     79 # if AppendSegmentRecording, get artifact times for each segment
   (...)
     91 # else:
     92 #     artifact_removed_valid_times, artifact_times = _get_artifact_times(recording, **artifact_params)
     94 key['artifact_times'] = artifact_times

File ~/Src/spyglass/src/spyglass/spikesorting/spikesorting_artifact.py:166, in _get_artifact_times(recording, zscore_thresh, amplitude_thresh, proportion_above_thresh, removal_window_ms)
    163 valid_timestamps = SpikeSortingRecording._get_recording_timestamps(
    164     recording)
    165 if recording.get_num_segments() > 1:
--> 166     recording = si.concatenate_recordings(recording.recording_list)
    168 # if both thresholds are None, we essentially skip artifract detection and
    169 # return an array with the times of the first and last samples of the recording
    170 if (amplitude_thresh is None) and (zscore_thresh is None):

AttributeError: 'BinaryRecordingExtractor' object has no attribute 'recording_list'
```

jguides, 2022-04-11:
It appears that the attribute _recording_segments is a substitute for the attribute recording_list.

samuelgarcia, 2022-04-11:
Hi,
in the new API, handle multi segment is quite more flexible.
Please have a deep look at:
Pleas ehave a look to this : https://spikeinterface.readthedocs.io/en/latest/modules/core/plot_5_append_concatenate_segments.html
and maybe check internal implementation. Theses list should be manipulated directly.

jguides, 2022-04-12:
Thanks @samuelgarcia. It looks like the first link was not included; could you post that again? I actually am getting a new error when I try to concatenate multiple recording segments. I will also note that prior to updating my spikeinterface, I was able to run similar lines without error. I had a look at the second link, but am unclear on how I can avoid the below error. Is it not possible to concatenate recordings that are of the BinaryRecordingSegment type? Thanks in advance for any help. 

`---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Input In [25], in <cell line: 11>()
      9 recording_name = SpikeSortingRecording._get_recording_name(key)
     10 # recording = si.load_extractor(recording_path)
---> 11 recording = si.concatenate_recordings(recording._recording_segments)

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/spikeinterface/core/segmentutils.py:218, in concatenate_recordings(*args, **kwargs)
    217 def concatenate_recordings(*args, **kwargs):
--> 218     return ConcatenateSegmentRecording(*args, **kwargs)

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/spikeinterface/core/segmentutils.py:129, in ConcatenateSegmentRecording.__init__(self, recording_list, ignore_times, sampling_frequency_max_diff)
    127 def __init__(self, recording_list, ignore_times=True, sampling_frequency_max_diff=0):
--> 129     one_rec = append_recordings(recording_list, sampling_frequency_max_diff=sampling_frequency_max_diff)
    131     BaseRecording.__init__(self, one_rec.get_sampling_frequency(), one_rec.channel_ids, one_rec.get_dtype())
    132     one_rec.copy_metadata(self)

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/spikeinterface/core/segmentutils.py:97, in append_recordings(*args, **kwargs)
     96 def append_recordings(*args, **kwargs):
---> 97     return AppendSegmentRecording(*args, **kwargs)

File ~/anaconda3/envs/spyglass/lib/python3.8/site-packages/spikeinterface/core/segmentutils.py:58, in AppendSegmentRecording.__init__(self, recording_list, sampling_frequency_max_diff)
     55 def __init__(self, recording_list, sampling_frequency_max_diff=0):
     57     rec0 = recording_list[0]
---> 58     sampling_frequency = rec0.get_sampling_frequency()
     59     dtype = rec0.get_dtype()
     60     channel_ids = rec0.channel_ids

AttributeError: 'BinaryRecordingSegment' object has no attribute 'get_sampling_frequency'`

alejoe91, 2022-04-12:
Hi @jguides 

Segments and recordings are two different concepts. Segments are sub parts of the same recording and treated very differently internally. The `recording._recording_segments` is private and should not be accessed directly. For the concatenation you can try: `si.concatenate_recordings([recording])`

Also I see that the error is triggered by a separate package. Can you send us or point us to the `spikesorting_artifact.py`?

jguides, 2022-04-12:
Thanks @alejoe91. This is helpful information. Also, this command worked for concatenating the segments of a recording (which I believe is the intent; I will double check with @khl02007 and @lfrank). The link to spikesorting_artifact.py is here: https://github.com/LorenFrankLab/spyglass/blob/master/src/spyglass/spikesorting/spikesorting_artifact.py.

closed 2022-04-12 by jguides