SpikeInterface/spikeinterface #601: BaseSorting fails to get spike trains if any units have only a single spike.

grahamfindlay, 2022-04-16:
First: thanks for the wonderful project. 
Second: Line 193 of `get_all_spike_trains` in the `BaseSorting` class (see below) fails to concatenate the list `spike_times` and raises an exception if you are unlucky enough to have any units with only 1 spike detected. This is because the single spike's timestamp is returned by `get_unit_spike_trains` as a 0-dimensional numpy array (scalar), rather than a 1-d numpy array of size 1. The fix is probably to just use `np.atleast_1d` instead of `np.array` wherever these timestamps are first read from disk. 

If you want, I can try to figure it out and submit a PR, or maybe someone who knows the codebase better than me can fix it in a quick second :) Just let me know.   

https://github.com/SpikeInterface/spikeinterface/blob/3788d84969fc22a76b5392126463812c4936af8b/spikeinterface/core/basesorting.py#L176-L199

alejoe91, 2022-04-20:
Hi @grahamfindlay 

Thanks for spotting this! If you have time and you want to dig into the code a bit more, it would be great if you could submit a PR! It seems you already know where to act exactly. 

Let us know!

Cheers
Alessio

TomBugnon, 2022-04-23:
Hi Alessio, `get_unit_spike_trains` appears in ~40 times in the repo, both BaseSorter, BaseSorterSegment and pretty much all individual sorting extractors. I gather all of those ought to be edited to call `np.atlease_1d` ? (Or is there a couple places downstream that are sufficient)

alejoe91, 2022-05-03:
@grahamfindlay is it using kilosort right?

grahamfindlay, 2022-05-04:
Hi, yes it is. Kilosort 2.5. Sorry I couldn't get to this in time, I see you're on it now though. Thanks!

alejoe91, 2022-05-05:
Indeed it should be fixed with https://github.com/SpikeInterface/spikeinterface/pull/617 (and it appears to be a Kilosort-specific problem)