SpikeInterface/spikeinterface #602: How to use plot_unit_waveforms API?

Yuhao-Zhang-SV, 2022-04-20:
Hi developers,

Sorry I have a simple question... After I read the API doc, there are two parameters, namely, _channel_ids_ and _units_ids_ in the _plot_unit_waveforms_ function. So basically can I just arbitrarily choose some units and channels as lists as inputs? I tried on this, but got **ValueError: operands could not be broadcast together with shapes (272,120,49) (1,1,2)** error. It seems that only when I choose **all** channels (not specify _channel_ids_ parameter). So is there a way that I can plot any number of units and channels as I want? Thanks in advance~

alejoe91, 2022-04-20:
Hi @Yuhao-Zhang-SV 

It would be helpful if you shared the snippet of code that you used! :)

Yuhao-Zhang-SV, 2022-04-21:
OK, I think this is how I called _plot_unit_waveforms_ API:

            self.we = si.extract_waveforms(self.loaded_recording_cmr, self.sorting_MS4, self.waveformCachePath, ms_before = 2, ms_after = 2, max_spikes_per_unit = 10000, overwrite=True)
            sw.plot_unit_waveforms(self.we, _channel_ids=[0, 2],_ unit_ids=[19, 37, 27, 29])

If I deleted channel_ids here, it could be executed successfully. (We have 49 channels, so I don't think there's an index error...) Thanks in advance~

alejoe91, 2022-04-21:
Hi @Yuhao-Zhang-SV 

Why are you using `_channel_ids` and `_unit_ids`? It should be `channel_ids` and `unit_ids` (without the `_`)!

alejoe91, 2022-04-21:
Also, can you paste the entire traceback?

Yuhao-Zhang-SV, 2022-04-21:
Oh, sorry about that... I intended to use the "_Italian_" format, but it did not display properly (the initial code has no _ before parameters...). And the entire traceback is shown below. Thanks~
[image]

alejoe91, 2022-04-21:
Ok thanks for the clarification! Can you try instead of providing `channel_ids` to use `radius_um=100`? Just to see if it works! This selects the channels in a um radius around the maximum channel

Yuhao-Zhang-SV, 2022-04-21:
OK, I think it works! The script can execute without errors. So why is this useful? I notice something has changed to my waveform figure...
[image]

alejoe91, 2022-04-21:
this plots the best channels for each unit. Also the `channel_ids` approach should work..I'll double check!

Yuhao-Zhang-SV, 2022-04-22:
Hi, since each unit has 9 waveforms, I just tried to include 9 channels to try to plot again but still failed... So what might be the problem here? Thanks in advance~
[image]

Yuhao-Zhang-SV, 2022-04-22:
Also, plot_unit_templates API has the same issue here...

alejoe91, 2022-04-22:
thanks for the report! I'll be off next week but we'll try to look into it at the beginning of May!

alejoe91, 2022-10-13:
HI @Yuhao-Zhang-SV 

We refactored the plotting functions completely. Can you check if the error is still there after upgrading? 
Closing for now, feel free to reopen!

closed 2022-10-13 by alejoe91