SpikeInterface/spikeinterface #604: recording extractor t_start

ClaraBesserer, 2022-04-20:
Hello,
I have recently discovered that my ephys recordings do not start at the same time (I use OpenEphys and record signals from headstage, digital and analog). They start with some delay that is saved in a 'timestamps.npy' file in my data folder, the first timestamps indicating the t_start of my data.
I use the binary recording extractor to import my data in spikeinterface (the data is stored in 'continuous.dat' files). I've seen that I can specify a 't_starts' parameter when importing the data, but what is the use of this parameter later on? When I use 'get_traces', the t_start is not taken into account at all and my traces are misaligned (see attached pic).

[image]

Is there a way to align the traces using Spikeinterface or do I have to do it manually?

For ref, here is the code I used:

`
gain=0.00024414062       
channel_offset = 0
dtype=np.dtype(np.int16)
time_axis = 0

headstage = base_input_folder/ 'continuous/eCube_Server-100.0' / 'continuous.dat'
hs_timestamps = np.load(base_input_folder/ 'continuous/eCube_Server-100.0' /'timestamps.npy')
num_channels = 64
sampling_frequency = 25000
recording_hs = spikeinterface.core.binaryrecordingextractor.BinaryRecordingExtractor(headstage, num_chan=num_channels, 
                                     sampling_frequency=sampling_frequency, dtype=dtype, gain_to_uV=gain, 
                                     offset_to_uV=channel_offset, time_axis=time_axis, t_starts = [float(hs_timestamps[0])])    

digital = base_input_folder/ 'continuous/eCube_Server-102.0' / 'continuous.dat'
dig_timestamps = np.load(base_input_folder/ 'continuous/eCube_Server-102.0' /'timestamps.npy')
num_channels = 64
sampling_frequency = 25000
recording_dig = spikeinterface.core.binaryrecordingextractor.BinaryRecordingExtractor(digital, num_chan=num_channels, 
                                  sampling_frequency=sampling_frequency, dtype=dtype, gain_to_uV=gain, 
                                  offset_to_uV=channel_offset, time_axis=time_axis, t_starts = [float(dig_timestamps[0])])    

analogique = base_input_folder/ 'continuous/eCube_Server-101.0' / 'continuous.dat'
ana_timestamps = np.load(base_input_folder/ 'continuous/eCube_Server-101.0' /'timestamps.npy')
num_channels = 32
sampling_frequency = 25000
recording_ana = spikeinterface.core.binaryrecordingextractor.BinaryRecordingExtractor(analogique, num_chan=num_channels, 
                                  sampling_frequency=sampling_frequency, dtype=dtype, gain_to_uV=gain, 
                                  offset_to_uV=channel_offset, time_axis=time_axis, t_starts = [float(ana_timestamps[0])])    

fs = 25000
trace_snippet_hs = recording_hs.get_traces(start_frame=int(fs*0), end_frame=int(fs*10), channel_ids=[41], return_scaled =True)
trace_snippet_dig = recording_dig.get_traces(start_frame=int(fs*0), end_frame=int(fs*10), channel_ids=[0], return_scaled =True)
trace_snippet_ana = recording_ana.get_traces(start_frame=int(fs*0), end_frame=int(fs*10), channel_ids=[1], return_scaled =True)

plt.figure()
plt.plot(trace_snippet_hs, label = 'HS')
plt.plot(trace_snippet_dig, label = 'dig')
plt.plot(trace_snippet_ana, label = 'ana')
plt.legend()
`

Thanks in advance!

samuelgarcia, 2022-05-04:
Salut Clara,
some recording handle times vector in two way:
  * with a times vectors
  * with t_start and sampling_rate, in that case the time vector is generated.

You can have with: ````rec.get_times()```

So in your case if you want to align your signal you should do:
```

trace_snippet_hs = recording_hs.get_traces(start_frame=int(fs0), end_frame=int(fs10), channel_ids=[41], return_scaled =True)
time_hs = recording_hs.get_times()[fs0:fs10]
trace_snippet_dig = recording_dig.get_traces(start_frame=int(fs0), end_frame=int(fs10), channel_ids=[0], return_scaled =True)
time_dig = recording_dig.get_times()[fs0:fs10]
trace_snippet_ana = recording_ana.get_traces(start_frame=int(fs0), end_frame=int(fs10), channel_ids=[1], return_scaled =True)
time_ana = recording_ana.get_times()[fs0:fs10]

fig, ax = plt.subplots()
ax.plot(time_hs, trace_snippet_hs, label = 'HS')
plt.plot(time_dig, trace_snippet_dig, label = 'dig')
plt.plot(time_ana, trace_snippet_ana, label = 'ana')
plt.legend()
```

I hope it helps.

alejoe91, 2022-05-04:
@ClaraBesserer you can also use the `read_openephys()` function instead of loading the binary file directly!

alejoe91, 2022-05-04:
@samuelgarcia does neo loads the `timestamps.npy` for open ephys?

samuelgarcia, 2022-05-04:
i don't remember. i think neo read the first timestamp only. need to check

closed 2022-10-13 by samuelgarcia