SpikeInterface/spikeinterface #605: Loading channel locations in nwb recordings

khl02007, 2022-04-20:
@lfrank @magland @alejoe91 

summary: #552 didn't resolve our issue

issue: we store data from both tetrodes and probes in the same NWB file. The location of the channels in a tetrode is defined on the xz plane (e.g. (`rel_x`, `rel_y`, `rel_z`) = (6.25, 0.0,  6.25)). On the other hand, the probe channels are defined on the xy plane ((`rel_x`, `rel_y`, `rel_z`) =(1086.00, -1080.0, 0.00)). right now, when we load the nwb file as a recording object in spikeinterface, the tetrode channel locations are not returned correctly even after specifying the axes. For example, for a tetrode channel 0, `recording.get_channel_locations(channel_ids=[0], axes='xyz')` returns `array([[6.25, 0.  , 0.  ]])` rather than `array([[6.25, 0.  , 6.25  ]])`. 

I think in the future spikeinterface should move toward 3d definition of channel locations as @magland and others suggested.

alejoe91, 2022-10-13:
@khl02007 any update on this?

khl02007, 2022-10-16:
@alejoe91 I don't think this can be fixed unless spikeinterface (or probeinterface?) supports 3D channel location. But this isn't really an urgent issue so if you want to close it, go ahead.

alejoe91, 2022-10-17:
I think that the best solution would be to implement the `read_nwb()` function in probeinterface, so you could do something like this:
```
import probeinterface as pi
rec_tetrode = ...
rec_shank = ...

# something in probeinterface to select the right probe
probe_tetrode = pi.read_nwb('path-to-neb.nwb', probe="electrodes_0")
probe_shank = pi.read_nwb('path-to-neb.nwb', probe="electrodes_1")

rec_tetrode = rec_tetrode.set_probe(probe_tetrode)
rec_shank = rec_shank.set_probe(probe_shank)
```

Let's move the conversation to probeinterface!

closed 2022-10-17 by alejoe91