SpikeInterface/spikeinterface #606: Automatic curation

phanhuynh, 2022-04-21:
Hello all,
I'm trying to do automatic curation based on this video:
https://www.youtube.com/watch?v=fvKG_-xQ4D8
[image]
Where can I find some sample code for the latest version of spikeinterface?
Thank you!

alejoe91, 2022-04-22:
Hi @phanhuynh 

you can find the new way here: https://github.com/SpikeInterface/spikeinterface/blob/master/examples/getting_started/plot_getting_started.py#L220-L239

phanhuynh, 2022-04-25:
Exactly what I was looking for!  Thank you!

phanhuynh, 2022-04-26:
Quick follow up.  Any ideas about this error?
```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File ~\anaconda3\envs\siphy_env\lib\site-packages\pandas\core\indexes\base.py:3621, in Index.get_loc(self, key, method, tolerance)
   3620 try:
-> 3621     return self._engine.get_loc(casted_key)
   3622 except KeyError as err:

File ~\anaconda3\envs\siphy_env\lib\site-packages\pandas\_libs\index.pyx:136, in pandas._libs.index.IndexEngine.get_loc()

File ~\anaconda3\envs\siphy_env\lib\site-packages\pandas\_libs\index.pyx:163, in pandas._libs.index.IndexEngine.get_loc()

File pandas\_libs\hashtable_class_helper.pxi:5198, in pandas._libs.hashtable.PyObjectHashTable.get_item()

File pandas\_libs\hashtable_class_helper.pxi:5206, in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'presence_ratio'

The above exception was the direct cause of the following exception:

KeyError                                  Traceback (most recent call last)
Input In [79], in <module>
      1 keep_mask = ((metrics['isi_violations_rate'] < 0.5) 
      2              & (metrics['amplitude_cutoff'] < 0.1)
----> 3               & (metrics['presence_ratio'] > 0.9))

File ~\anaconda3\envs\siphy_env\lib\site-packages\pandas\core\frame.py:3505, in DataFrame.__getitem__(self, key)
   3503 if self.columns.nlevels > 1:
   3504     return self._getitem_multilevel(key)
-> 3505 indexer = self.columns.get_loc(key)
   3506 if is_integer(indexer):
   3507     indexer = [indexer]

File ~\anaconda3\envs\siphy_env\lib\site-packages\pandas\core\indexes\base.py:3623, in Index.get_loc(self, key, method, tolerance)
   3621     return self._engine.get_loc(casted_key)
   3622 except KeyError as err:
-> 3623     raise KeyError(key) from err
   3624 except TypeError:
   3625     # If we have a listlike key, _check_indexing_error will raise
   3626     #  InvalidIndexError. Otherwise we fall through and re-raise
   3627     #  the TypeError.
   3628     self._check_indexing_error(key)

KeyError: 'presence_ratio'
```

alejoe91, 2022-08-01:
Seems the `presence_ratio` was not included in the `metrics`

closed 2022-08-01 by alejoe91