SpikeInterface/spikeinterface #609: Little performance improvement

ferchaure, 2022-04-26:
Hi guys, I was coding the curationsort class and I started wondering, the output of get_unit_spike_train  must be sorted? 
If that is the case, we could use np.searchsorted

Code like this is O(log(n)):

`   init = np.searchsorted(times, start_frame, side='left')`
`   endf = np.searchsorted(times, end_frame, side='right')`
`   newtimes = times[init:endf]`

But the usual code is O(n):
 `newtimes = times[(times>start_frame) & (times<=end_frame)]`

Cheers
Fer

samuelgarcia, 2022-05-09:
Hi fernando.
Yes it must be sorted.
Go for the searchsorted approach it is faster.

ferchaure, 2022-05-09:
cool, I will try to remember it if I found the same codes in other places

closed 2022-05-09 by ferchaure