SpikeInterface/spikeinterface #610: Klusta units are not unique! Using unique unit ids

jkb521, 2022-04-27:
I use the Klusta sorts , but when finished,
I got  "Klusta units are not unique! Using unique unit ids"., and  did not generate files such as spike_times.npy ......,
and thus cannot compute quality_metrics analyses 

I have four groups 16 channels.

samuelgarcia, 2022-05-09:
@alejoe91 : can you help on this ?

alejoe91, 2022-05-09:
Hi @jkb521 

Can you share the entire code and error? `"Klusta units are not unique! Using unique unit ids"` is just a warning and it renames the unit ids.. the problem should not be related. Note that spike times are not in the file list for klusta, which generates a `kwik` file.

alejoe91, 2022-08-01:
This is not an error, but a warning that klusta units from different channel groups are renamed

closed 2022-08-01 by alejoe91