SpikeInterface/spikeinterface #615: `plot_rasters` gives error if spike_train contains a single spike

lkeegan, 2022-05-03:
```
TypeError                                 Traceback (most recent call last)
Input In [4], in <module>
      1 fig, ax = plt.subplots(figsize=(24, 8))
----> 2 w_rs = sw.plot_rasters(sorting, ax=ax)

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/widgets/rasters.py:107, in plot_rasters(*args, **kwargs)
    106 def plot_rasters(*args, **kwargs):
--> 107     W = RasterWidget(*args, **kwargs)
    108     W.plot()
    109     return W

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/widgets/rasters.py:55, in RasterWidget.__init__(self, sorting, segment_index, unit_ids, time_range, color, figure, ax)
     52 for unit_id in self._sorting.get_unit_ids():
     53     spike_train = self._sorting.get_unit_spike_train(unit_id,
     54                                                      segment_index=self.segment_index)
---> 55     if len(spike_train) > 0:
     56         curr_max_frame = np.max(spike_train)
     57         if curr_max_frame > self._max_frame:

TypeError: len() of unsized object
```
Related to #601 - but I think just using `.size` instead of `len()` here would fix this.

alejoe91, 2022-05-03:
Hi @lkeegan 

This came up before! Another **deeper** fix would be to ensure spiketrains are `np.atleast1d` in the `get_unit_spike_train` function. I think this would fix it too and it would probably also fix other bugs related to this elsewhere. 

Will make a quick PR now! :)

lkeegan, 2022-05-03:
Hi @alejoe91 - that would be great, thanks! I just ran into another similar issue which would likely also be fixed by using `np.atleast1d`:
```
IndexError                                Traceback (most recent call last)
Input In [9], in <module>
      1 # extract waveforms from sorting:
----> 2 si.extract_waveforms(saved_recording, sorting, folder= str(output_path / "wf_ks2"), 
      3                               max_spikes_per_unit=None, n_jobs=16, total_memory="1G", progress_bar=True, overwrite=True)

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py:879, in extract_waveforms(recording, sorting, folder, load_if_exists, precompute_template, ms_before, ms_after, max_spikes_per_unit, overwrite, return_scaled, dtype, use_relative_path, **job_kwargs)
    876 we = WaveformExtractor.create(recording, sorting, folder, use_relative_path=use_relative_path)
    877 we.set_params(ms_before=ms_before, ms_after=ms_after, max_spikes_per_unit=max_spikes_per_unit, dtype=dtype,
    878               return_scaled=return_scaled)
--> 879 we.run_extract_waveforms(**job_kwargs)
    881 if precompute_template is not None:
    882     we.precompute_templates(modes=precompute_template)

File /gpfs/bwfor/work/ws/hd_uk239-measelab/miniconda3/envs/measelab/lib/python3.8/site-packages/spikeinterface/core/waveform_extractor.py:661, in WaveformExtractor.run_extract_waveforms(self, **job_kwargs)
    659         spike_times = self.sorting.get_unit_spike_train(unit_id=unit_id, segment_index=segment_index)
    660         sel = selected_spikes[unit_id][segment_index]
--> 661         selected_spike_times[unit_id].append(spike_times[sel])
    663 # prepare memmap
    664 wfs_memmap = {}

IndexError: too many indices for array: array is 0-dimensional, but 1 were indexed
```

alejoe91, 2022-05-03:
@lkeegan can you check if this PR https://github.com/SpikeInterface/spikeinterface/pull/617 fixes the issue? I think it would be a more general fix!

lkeegan, 2022-05-03:
@alejoe91 thanks! that PR fixes both issues for me

alejoe91, 2022-05-03:
We'll merge that then!

alejoe91, 2022-05-03:
@lkeegan is this using kilosort right?

lkeegan, 2022-05-03:
@alejoe91 yes, this was using kilosort 2

alejoe91, 2022-05-03:
I'm asking because it the problem is specifically due to KS then it's better to change its extractor directly. I'll update the PR. Let me know if the problem is still fixed :)

closed 2022-05-03 by samuelgarcia