SpikeInterface/spikeinterface #639: Question: Best practices for sorting spike snippets, rather than continuous recordings

grahamfindlay, 2022-05-25:
Although we now collect all our spiking data as continuous recordings, we have a fair amount of old data where threshold crossings were detected online, and short snippets (a few ms) around the crossing were extracted. This was done to reduce storage requirements for long (>24h) recordings, back when such things were prohibitively expensive. These data were all collected on TDT systems, which actually had a dedicated data structure (called "snips", I think) for this sort of thing, because it was fairly common. 

Is there a canonical way using SpikeInterface to ingest this sort of data and pass it to sorters that support it (e.g. WaveClus, which we used to use for snippet data)? 

I have seen several examples of people trying to artificially construct continuous recordings from snippets, for example by zero-padding around the snippets, or just concatenating them all together and doing bookkeeping to map returned spike times to snippet times, but I actually can't recall anyone reporting satisfactory results with these methods, and they seem cumbersome. Here's one example: [link](https://github.com/flatironinstitute/mountainsort/issues/16). 

This may already be possible, I just couldn't find a clear answer from the docs/examples. Thanks!

ferchaure, 2022-05-25:
Implement clustering inside spikeinterface is one of the projects on the [hackathon ](https://catalystneuro.github.io/spike-sorting-hackathon/), and already extracted spikes can be seen as the output of  the clustering first steps. Then maybe the extracted spikes/snippets can be the same class or at least that's my idea for fixing the issue in the hackathon.

alejoe91, 2022-05-26:
Thanks for the question @grahamfindlay 

As @ferchaure mentioned currently we do not have a way to handle this. 

In `spikeextractors`, we had an example of a `RecordingExtractor` from Axona that fills in the missing portions of the signals with low-amplitude gaussian noise (see [here](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/axonaunitrecordingextractor/axonaunitrecordingextractor.py))

This could also be an option and we successfully ran several sorters on these "Frankestein" traces. However, I believe that what Fernando suggests is a better approach and we should discuss about that. 

Also feel free to sign up for the hackathon that Fernando mentioned, so we can make some progress in that direction.

Cheers
Alessio

alejoe91, 2022-08-01:
@ferchaure implemented the `SnippetExtractor` and `WaveClus` now supports running on snippets data: https://github.com/SpikeInterface/spikeinterface/pull/764

closed 2022-08-01 by alejoe91