SpikeInterface/spikeinterface #642: Load sorting from mountainsort output folder

mfvd, 2022-05-26:
Hello,

I've been trying to read a sorting object from the mountainsort output folder. I was expecting to find something similar to ```se.read_kilosort("kilosort-folder")``` but don't seem to find the correct method. 

This dataset was sorted by property (shank) and the output folder has one sub-folder for each shank.

How can I load the this data into SI so I can export to phy?

Cheers

alejoe91, 2022-05-26:
Hi @mfvd 

Mountainsort uses the MDA format, so you can use these functions: https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/extractors/mdaextractors.py#L215-L236

mfvd, 2022-05-27:
Hi @alejoe91 

Thanks.
I've tried `sorted_ms4 = se.read_mda_sorting('ms4_output')` but got the following error

```python
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [13], in <cell line: 3>()
      1 #params= dict 
      2 #params['sampling_frequency'] = 30000
----> 3 sorted_ms4 = se.read_mda_sorting('ms4_output')

File /opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/extractors/mdaextractors.py:221, in read_mda_sorting(file_path, **kwargs)
    220 def read_mda_sorting(file_path, **kwargs):
--> 221     sorting = MdaSortingExtractor(file_path, **kwargs)
    222     return sorting

TypeError: __init__() missing 1 required positional argument: 'sampling_frequency'
```

alejoe91, 2022-05-27:
Yeah you gave to pass the sampling frequency as an additional arg because mda doesn't have it

mfvd, 2022-05-27:
My bad, I assumed it was in the sorting object.

However, I get this new error now

```python
---------------------------------------------------------------------------
IsADirectoryError                         Traceback (most recent call last)
Input In [14], in <cell line: 3>()
      1 #params= dict 
      2 #params['sampling_frequency'] = 30000
----> 3 sorted_ms4 = se.read_mda_sorting('ms4_output', sampling_frequency=30000)

File /opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/extractors/mdaextractors.py:221, in read_mda_sorting(file_path, **kwargs)
    220 def read_mda_sorting(file_path, **kwargs):
--> 221     sorting = MdaSortingExtractor(file_path, **kwargs)
    222     return sorting

File /opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/extractors/mdaextractors.py:145, in MdaSortingExtractor.__init__(self, file_path, sampling_frequency)
    144 def __init__(self, file_path, sampling_frequency):
--> 145     firings = readmda(str(file_path))
    146     labels = firings[2, :]
    147     unit_ids = np.unique(labels).astype(int)

File /opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/extractors/mdaextractors.py:545, in readmda(path)
    543 if file_extension(path) == '.npy':
    544     return readnpy(path);
--> 545 H = _read_header(path)
    546 if H is None:
    547     print("Problem reading header of: {}".format(path))

File /opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/extractors/mdaextractors.py:414, in _read_header(path)
    411     Path(tmp_fname).unlink()
    412     return ret
--> 414 f = open(path, "rb")
    415 try:
    416     dt_code = _read_int32(f)

IsADirectoryError: [Errno 21] Is a directory: 'ms4_output'
```

alejoe91, 2022-05-27:
Ah you have to point to the actual .mda file for the spiking data

mfvd, 2022-05-27:
Ahh ok, that makes sense! 
Would there be a way to load all sorting objects (one per shank) as a single object. My goal would be to have all the data exported to phy as a single object.

alejoe91, 2022-05-27:
You can loop over the folders and load the individual sorting in a list and then use the `si.aggregate_units(sorting_list)` function (see [here](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/core/unitsaggregationsorting.py#L100))

mfvd, 2022-05-27:
Thanks @alejoe91! I'll do that.

cheers

closed 2022-05-27 by mfvd