SpikeInterface/spikeinterface #644: extra dependencies for sorting in containers

JuliaSprenger, 2022-05-27:
Depending on the input format used for running a sorter in a container some extra requirements need to be installed, e.g. nixio (for more potential additional requirements see [here](https://github.com/NeuralEnsemble/python-neo/blob/2b0550ec8c4742dcfffcbe5f900027f97026ec9f/setup.py#L9). These are currently neither installed by default, nor installed when an upgrade of spikeinterface is performed in the container.
Should these extra dependencies be part of the default installation, installed depending on the input format or would it be best to add an `extras_require` argument to the `run_sorter_container` function to be specified by the user?

alejoe91, 2022-05-27:
Hi @JuliaSprenger 

I think that we can give both options: installing some additional packages with `extra_requires` and having some class variables for extractors that require additional packages, so we can recursively look into the recording dictionary to see whether some additional.packages are needed. What do you think?

samuelgarcia, 2022-05-30:
@JuliaSprenger : honestly I would build a docker image based on the one you want to run with nixio preinstalled!

JuliaSprenger, 2022-05-30:
Hi @samuelgarcia Yes, that's an option for me to build my own container, but I don't think it would be very user friendly to provided official containers that only work for some formats, as not all users are familiar with containerization and package installation.

alejoe91, 2022-06-13:
Fixed via https://github.com/SpikeInterface/spikeinterface/pull/689

closed 2022-06-13 by alejoe91