SpikeInterface/spikeinterface #645: MEArecRecordingExtractor.get_traces() return array instead of memmap

Dradeliomecus, 2022-05-27:
Hi,

Working with the `MEArecRecordingExtractor`, I found out that using the `get_traces()` method returns a numpy array rather than a `memmap`.
This means that filtering the recording leads to the program getting killed because there is not enough RAM (and without filtering, I can barely get the traces (~40GiB).

Is this by design?

Thanks,
Aurélien W.

alejoe91, 2022-05-27:
Hi Aurelien, 

The `get_traces` on the Mearec object should return an h5 view. However, when filtering you have to load the traces in RAM. You want be able to filter in memory, but you can save the object using parallel chunks that write to a binary file.

Dradeliomecus, 2022-05-27:
> The `get_traces` on the Mearec object should return an h5 view.
Here is what I get:
```python
import spikeinterface.extractors as se

recording = se.MEArecRecordingExtractor("/path/to/recording_neuropixels_30m_200cells.h5")
print(type(recording.get_traces()))
  <class 'numpy.ndarray'>
```

> you can save the object using parallel chunks that write to a binary file.
How do I do that?

Thank you!

alejoe91, 2022-05-27:
Ok I'll look into the memmap issue. For saving, you can use:
`rec_bin = recording.save(folder=..., chunk_duration='1s', n_jobs=4, progress_bar=True)`

Also note that you shouldn't really call the get traces of an object with no arguments, but always slice it using the arguments (`start_frame`, `end_frame`, `channel_ids`). Get traces is supposed to return you the actual traces, and most of the times they are loaded in mem

Dradeliomecus, 2022-05-27:
> Also note that you shouldn't really call the get traces of an object with no arguments
Yeah I usually only do that for debugging, but here I'm pointing that out because I get a crash when loading the waveforms with a filtered recording (because of not enough RAM), and I believe that the problem comes from the get_traces() that uses RAM instead of disk.

This means that in order to get the filtered waveforms, I need to first save the recording to a binary file on disk and extract the waveforms on this new recording object?

samuelgarcia, 2022-05-30:
@Dradeliomecus : `get_traces()` is by design the **load in memory** operation.
Having a numpy array is the correct behavior.
You can also have a numpy.memmap in some case but it is still a numpy array.
When it is h5 based then the h5 view is converted to numpy array on the fly.

No :  to get the filetered waveforms you do not need to save (cache) the entired recording.
You can used the lasy recording object for that.
`extract_waveforms()` have a mechanism to load on the fly small chunk and this should not be memory agressive.

In many place we use internally `ChunkRecordingExecutor` (see job_tools.py) which have the ability to load small chunk in parallel with **job_kwargs** (for instance `chunk_duration='1s', n_jobs=4, progress_bar=True`) that define how much memory and how many process you want you want for a operation (`extract_waveforms()`  or `save()`)

Dradeliomecus, 2022-05-30:
Thank you @samuelgarcia for the clarification!

I see, I did not see anything in the source code about chunking, but it actually was in the `**kwargs`
https://github.com/SpikeInterface/spikeinterface/blob/1c9c9efecaf2a88111eeda9443ca2aabaafb391c/spikeinterface/core/waveform_extractor.py#L736

closed 2022-05-30 by Dradeliomecus