SpikeInterface/spikeinterface #652: Kilosort hidden parameters

maurotoro, 2022-06-03:
Dearest,

During my sorting adventures, I realized that kilosort has a parameter that controls the length of the waveform that is taken from the recording that is not mentioned on their StandardConfig.m files. From my experience, changing their default value of 60 to 30 improve tremendously my results, so I would like to be able to change it directly through spikeInterface. Currently, I have to make the pipeline with SI, and before it runs the sorter, I add a hack to add it on the resulting `kilosort_config.m`.

Can I make a pull request to have access to this `ops.nt0` parameter from SI, pretty please??

Many thanks for the library, and all the amazing work,

Suerte!

alejoe91, 2022-06-03:
Hola @maurotoro 

Of course! Can you make a PR to expose the parameter? As you might have seen,we are currently refactoring the KS implementations, so please make it as soon as you can so we can propagate the changes ;)

Cheers
Alessio

alejoe91, 2022-06-13:
Fixed via https://github.com/SpikeInterface/spikeinterface/pull/655

closed 2022-06-13 by alejoe91