SpikeInterface/spikeinterface #657: Spike sorting Intan data with Kilosort gives weird results due to casting

johnmbarrett, 2022-06-07:
Hi,

First off, apologies if this has been brought up before, but I couldn't find this exact issue when searching for issues relating to Intan or Kilosort (and none when searching for Intan and Kilosort). Also not sure if this is an issue with this package or Neo, if the latter please let me know and I'll bring it up there.

Anywho, when trying to use SpikeInterface to run some Intan .rhd files through Kilosort, I was able to get it to run, but the extracted waveforms were incredibly weird, with a sort of high-frequency square wave shape. After some digging, I think this is caused by improper casting: Intan .rhd data is natively stored as uint16 with 0V = 2^15 (and this is also how Neo extracts it), so assuming your signals oscillate around zero, then the high bit of each sample will switch fairly rapidly between 0 and 1. However, when Kilosort3Sorter prepares its data, it casts the traces to int16, so instead of oscillating around zero, the samples rapidly switch between +2^15 and -2^15. Then when Kilosort reads it back in also as int16, it effectively gets corrupted data.

Steps to reproduce:

1) Download the [image] and rename it to have the extension .py.
2) Download this example .rhd file: https://drive.google.com/file/d/10C4VRPuFPjoCxkNtlsgr4LYM6j0bJYLQ/view?usp=sharing
3) Change the paths on lines 9, 20, 24 to point to the location your downloaded the data, your Matlab path, and your Kilosort path, respectively
4) Run the script

Expected results: vaguely sensible spike sorting

Actual results: weird waveforms with large jumps

Additional info:

I've attached plots before and after the cast on line 179 of core_tools.py to illustrate the issue. Pretty sure I can whip up a workaround that will work for me, but not sure what the cleanest solution is within the overall design of spikeinterface, hence my opening an issue rather than trying to fix it myself.

Before cast:
[image]

After cast:
[image]

johnmbarrett, 2022-06-07:
Version info, since I forgot to mention it:

spikeinterface: 0.94.0
Python: 3.9.6

$ uname -a
Linux phys-narwhal 4.15.0-123-generic #126-Ubuntu SMP Wed Oct 21 09:40:11 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Also looks like some recent commits (since the latest version in pip) have pushed the cast in question down to line 180 of core_tools.py

johnmbarrett, 2022-07-07:
Hi, apologies for the slow response. No, this happens before extracting waveforms step when setting up the folder for spike sorting. The recording.dat that is fed to Kilosort is corrupted by the casting. To see what I mean, do the following:

1) Follow the steps in my original post
2) Kill the Matlab process before it finishes spike sorting
3) Start a new Matlab instance and navigate to the temp folder created on line 27 of my script (it should be printed to standard output of whatever environment is running the script)
4) Set a breakpoint on line 85 of preprocessDataSub.m (`if isempty(buff)`)
5) Run kilosort3_master.m from the temp folder
6) Type `figure; plot(buff')` in the Matlab command window.

You should get a figure like the attached. This is the actual data that gets fed into Kilosort, which then leads to incorrect spike sorting (because of the random +/- 6 mV jumps interspersed throughout every trace).

[image]

samuelgarcia, 2022-07-07:
uint16 play not well generally with sorters.
A good trick is to cast then to float32 or int16.
This can be done at the filetring stage with this:
```python
rec_f = si.bandpass_filter(rec, dtype='int16')
sorting = si.run_sorter('kilosortX', rec_f, output_folder='/my/working/path')
```

In that case, rec_f will be filetred and then cast to int16 on the fly.
The drawback is that you will have temporary copy in your working folder.

johnmbarrett, 2022-07-07:
What if I don't want to filter the recording beforehand? Kilosort already has a filtering step. More generally, why should user need to know that Intan RHD files are internally stored as uint16? To my mind it would be better to abstract that sort of implementation detail behind the API, so the user can seamlessly mix and match different Extractors and Sorters, but if you disagree fair enough.

alejoe91, 2022-07-08:
Hi @johnmbarrett 

Thanks for the additional clarification! I agree this is an issue that should be solved on our side. A possible solution would be the following: when saving to binary in `int16` (or othe iteger types), we can check if the input data is `uint` and in that case we can make to offset and casting on the fly (similar to what we do with filtering, see [here](https://github.com/SpikeInterface/spikeinterface/blob/2b53b3c11b739e86f2d132f2c9ad4bdc15151c54/spikeinterface/preprocessing/filter.py#L114-L115), but in this case actually applying the `2^(n-1)` offset before casting to int).

Dows that sound good? I'll start drafting a PR!

closed 2022-07-08 by samuelgarcia

johnmbarrett, 2022-07-08:
Looks good, just upgraded to the new version and ran my test script and all my spikes come out looking like spikes! Thanks for all your help