SpikeInterface/spikeinterface #708: Calling variable after re-opening Jupyter (without dill.dump_session)

phanhuynh, 2022-06-15:
Hello everyone,
is there a way to call a variable once I've closed a jupyter session? What happened was that I ran the sorter:
sorting_KS3 = ss.run_kilosort3(recording_saved, output_folder= '_output_KS3'...)
and it successfully created the _output_KS3 folder.
But I did not run dill.dump_session(...)
Now, I've reopened Jupyter and I want to call sorting_KS3 without running the sorter again. Thoughts?

samuelgarcia, 2022-06-16:
You can do 
`sorting_KS3.save('/path/to/a/folder')`
and in a new session

`sorting_KS3_saved = si.load_extractor('/path/to/a/folder')`

sorting_KS3_saved will do not depend anymore on the '_output_KS3' folder then.

alejoe91, 2022-06-16:
alternative solution here: https://github.com/SpikeInterface/spikeinterface/issues/702

closed 2022-06-16 by alejoe91