SpikeInterface/spikeinterface #716: Python.exe - Entry Point Not Found

phanhuynh, 2022-06-16:
Hello everyone,
I've been getting this error message:
[image]

It pops up when I execute different lines of code, including:
```
mcmp = sc.compare_multiple_sorters([sorting_KS3, sorting_IC], ['KS', 'IC'], 
                                   spiketrain_mode='union', verbose=True)
```
I've tried:
-reinstalling anacoda
-reinstalling jupyter
-recreating the env

Any thoughts?

alejoe91, 2022-06-16:
I think that the problem is related to multiprocessing. Can you try to run the function with `n_jobs=1`?

phanhuynh, 2022-06-20:
Thank you so much!  It seems to be working...but does this mean a slower process?

alejoe91, 2022-06-20:
Yes, but not that much ;) you can try to reinstall joblib and refer to the joblib docs for windows if you find the same error

phanhuynh, 2022-06-20:
The problem did not go away unfortunately.  The error window pops up as soon as I open Jupyter.  
Oddly, I can close the error window a few times, and things seems to work again.  But the window will continue to appear randomly (without crashing things).  

I'm concerned that there are problems I can't see.  At the least, it's rather annoying.

I'm not sure how to follow your instruction of "refer to the joblib docs for windows".  

I'm considering the solution offered here:
https://stackoverflow.com/questions/57254007/how-to-fix-entry-point-not-found-while-installing-libraries-in-conda-environment

Thoughts?

alejoe91, 2022-06-20:
Nothing comes to mind and unfortunately we are not Windows users..the solution that you shared seems promising, let us know how it goes!

alejoe91, 2022-08-01:
Related to conda+joblib rather than SI