SpikeInterface/spikeinterface #718: Potential instability in spatial whitening

miketrumpis, 2022-06-16:
Applied naively, this singular value reciprocal can potentially blow up.

https://github.com/SpikeInterface/spikeinterface/blob/48580e9b44725397964055818d3d5338cf822309/spikeinterface/toolkit/preprocessing/whiten.py#L35-L36

Perhaps use a relative tolerance? For paranoia, I threw in some outlier checking on the random chunks. For good measure, There is the option to skip spatial normalization.

```python
    def __init__(self, recording, spatial: bool=True, rtol: float=1e-6, **random_chunk_kwargs):
        random_chunk_kwargs['concatenated'] = False
        random_data = get_random_data_chunks(recording, **random_chunk_kwargs)
        chunk_var = [b.var() for b in random_data]
        q_lo, q_hi = np.percentile(chunk_var, [25, 75])
        thresh = q_hi + 1.5 * (q_hi - q_lo)
        good_blocks = [b for b, v in zip(random_data, chunk_var) if v < thresh]
        random_data = np.concatenate(good_blocks, axis=0)

        # compute whitening matrix
        M = np.mean(random_data, axis=0)
        M = M[None, :]
        data = random_data - M
        if spatial:
            AAt = data.T @ data
            AAt = AAt / data.shape[0]
            U, S, Ut = np.linalg.svd(AAt, full_matrices=True)
            n = len(S[S > rtol * S[0]])
            W = (U[:, :n] @ np.diag(1 / np.sqrt(S[:n]))) @ Ut[:n]
        else:
            channel_sds = np.array([b.std(axis=0) for b in good_blocks])
            median_sd = np.median(channel_sds, axis=0)
            W = np.diag(1 / median_sd)
```

Apologies this is in a not-forked monkey patch and not PR-able just now.

samuelgarcia, 2022-06-17:
Hi,
thanks for the feedback.
Personnay, I don't like and don't use this whitening preprocessing. I prefer also by channel normallalisation (no spatial one).
For this you can use si.toolkit.zscore wich do more or less the same but without the trick of block selection but use median/mad for the diagonal.
The actual implementation in SI is here because some user use it.
I so know that the implementation IN SI is dangerous I wil try to improve this with your trick.

miketrumpis, 2022-06-17:
Aha! Yeah I was digging into this processor because it's kind of secretly used in the mountainsort object. I much prefer what you pointed out. There's so much here that it can be hard to find it :)

alejoe91, 2022-06-17:
In our defense, this was a last-week addition ;)

closed 2022-08-02 by alejoe91