SpikeInterface/spikeinterface #720: KeyError for ironclust and kilosort in container mode

chyumin, 2022-06-18:
`KeyError` happens when running ironclust or kilosort without gpu parameter:

```python
sorting = ss.run_sorter(
    'ironclust',  # or 'kilosort'
    recording,
    singularity_image=True,
    verbose=True,
    delete_output_folder=False,
)
```

For kilosort:
```
KeyError: 'useGPU'
```

For ironclust:
```
KeyError: 'fGpu'
```

I debugged a little bit:

`sorter_params` in this [line](https://github.com/SpikeInterface/spikeinterface/blob/901d66479cd4a6171e9733ad1b57c451d09b0bab/spikeinterface/sorters/runsorter.py#L409) is an empty dict

chyumin, 2022-06-18:
Runs well when adding the sorter parameter

alejoe91, 2022-06-18:
Thanks for spotting this @chyumin 

We should merge the `**sorter_params` and the default params dict before then. Can you make a PR?

samuelgarcia, 2022-06-20:
The merging of params (given ones  + default ones) is done in `Sorter.set_params()`
We should stick with that.
The GPU check is done before in the container pipeline.
I would rather consolidate `Sorter.use_gpu(sorter_params)` to accept empty dict.

alejoe91, 2022-06-21:
Fixed via https://github.com/SpikeInterface/spikeinterface/pull/722

closed 2022-06-21 by alejoe91