SpikeInterface/spikeinterface #736: TypeError: cannot pickle 'weakref' object

phanhuynh, 2022-06-23:
Hello dream-team, any thoughts on this error?  Note that despite the error, the sorter executed successfully.  
I've attached the file log but it doesn't explain a lot.  
[spikeinterface_log.txt](https://github.com/SpikeInterface/spikeinterface/files/8966883/spikeinterface_log.txt)
```
Docker: pulling image spikeinterface/spyking-circus-base
Starting container
Installing spikeinterface from sources in spikeinterface/spyking-circus-base
Installing dev spikeinterface from remote repository
Installing extra requirements: ['neo']
Running spykingcircus sorter inside spikeinterface/spyking-circus-base
Stopping container
--- _output_SC took 1422.3418962955475 seconds ---
Spyking found 269 units
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [38], in <cell line: 21>()

     19 print("--- _output_SC took %s seconds ---" % (time.time() - start_time))
     20 print(f'Spyking found {len(sorting_SC.get_unit_ids())} units')
---> 21 get_ipython().run_line_magic('store', 'sorting_SC')

File ~\Anaconda3\envs\si_env\lib\site-packages\IPython\core\interactiveshell.py:2305, in InteractiveShell.run_line_magic(self, magic_name, line, _stack_depth)
   2303     kwargs['local_ns'] = self.get_local_scope(stack_depth)
   2304 with self.builtin_trap:
-> 2305     result = fn(*args, **kwargs)
   2306 return result

File ~\Anaconda3\envs\si_env\lib\site-packages\IPython\extensions\storemagic.py:229, in StoreMagics.store(self, parameter_s)
    227     return
    228 #pickled = pickle.dumps(obj)
--> 229 db[ 'autorestore/' + arg ] = obj
    230 print("Stored '%s' (%s)" % (arg, obj.__class__.__name__))

File ~\Anaconda3\envs\si_env\lib\site-packages\pickleshare.py:118, in PickleShareDB.__setitem__(self, key, value)
    115 # We specify protocol 2, so that we can mostly go between Python 2
    116 # and Python 3. We can upgrade to protocol 3 when Python 2 is obsolete.
    117 with fil.open('wb') as f:
--> 118     pickle.dump(value, f, protocol=2)
    119 try:
    120     self.cache[fil] = (value, fil.stat().st_mtime)

TypeError: cannot pickle 'weakref' object
```

phanhuynh, 2022-06-23:
I see!  Thank you.  So,
 
To reload from ironclust:
`sorting_IC = ss.sorter_dict['ironclust'].get_result_from_folder('C:/Users/huynh/spike_epsztein/proj_louis/_output_IC')
`

From kilosort3:
`sorting_KS3 = se.read_kilosort('C:/Users/huynh/spike_epsztein/proj_louis/_output_KS3')
`

BUT, from spyking circus:
`sorting_SC = ss.sorter_dict['ironclust'].get_result_from_folder('C:/Users/huynh/spike_epsztein/proj_louis/_output_IC')
`gives me the error: 
`ValueError: This folder is not a cached folder C:\Users\huynh\spike_epsztein\proj_louis_mini\_output_SC
`

Also, since it's not consistent, what is it for tridesclous?  

Thank you again for all your help :) !

alejoe91, 2022-06-23:
Can you paste the code? You have `ironclust` instead of `spykingcircus` there. In theory that should work.

I recommend to save the output in a compact format with the `save()` function:
`sorting_SC = sorting_SC.save(folder="SC")`

Then you can reload with:
`sorting_SC = si.load_extractor("SC")`

this way you domn't have to keep track of all the output folders!

phanhuynh, 2022-06-23:
That is FANTASTIC.  I see that there's no option to remove_existing_folder=True, is that correct?

phanhuynh, 2022-06-29:
Hi @alejoe91 , it appears this doesn't work for this object? 

```
recording_saved = recording.save(folder="_preprocessed", progress_bar=True, 
                                     n_jobs=4, total_memory="100M")

recording_saved = recording_saved.save(folder="_preprocessed_backup")

recording_saved = si.load_extractor("_preprocessed_backup")
```

alejoe91, 2022-06-29:
What's the error?

phanhuynh, 2022-06-29:
Ah sorry, it works fine!  Somehow I got an error before about missing cache folder, but it works perfectly now :)

closed 2022-07-19 by alejoe91