SpikeInterface/spikeinterface #746: Error when pip install spikeinterface

TRuikes, 2022-06-28:
When I run pip install spikeinterface (on unix) I get:
```
(spikesorting_2022) [truikes@csn-h0 ~]$ pip install spikeinterface
Collecting spikeinterface
  Using cached https://files.pythonhosted.org/packages/13/67/4293464d741bbbf171de6fbb3d26565434fcd488d31698d70feb331b04d6/spikeinterface-0.94.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-GyMG6O/spikeinterface/setup.py", line 9, in <module>
        install_requires = open_requirements('requirements.txt')
      File "/tmp/pip-build-GyMG6O/spikeinterface/setup.py", line 4, in open_requirements
        with open(fname, mode='r') as f:
    IOError: [Errno 2] No such file or directory: 'requirements.txt'

alejoe91, 2022-06-28:
I tried to reproduce the issue but it's working on my side:

```
conda create -n si_test python=3.8

conda activate si_test

pip install spikeinterface
```
and everything is properly installed. Are you using conda?

TRuikes, 2022-06-28:
that works, the environment I was installing in was pyton 3.7. 
On the installation page its not mentioned at least python 3.8 is required:
https://spikeinterface.readthedocs.io/en/latest/installation.html

alejoe91, 2022-06-28:
You're right! We'll add it to the docs

TRuikes, 2022-06-28:
I just checked on my local PC on windows and there it works with python 3.7. In any case it works now